espressif/arduino-esp32 #2362: Arduino BluetoothSerial crash after streaming data for a period, especially on interrupt.

moothyknight, 2019-01-21:
### Hardware:
Board:							LOLIN32 V1.0.0
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                    ?
Upload Speed:					921600
Computer OS:                                        Win 10 and Android

### Description:
Describe your problem here
Am trying to sync a phone with a bluetooth serial stream. I've gotten the stream working in BLE but could not read the write bytes on the ESP32 correctly in the UART example when I tried writing to it. So I switched to SerialBT which is more simple for my needs, but now I'm getting periodic crashing after streaming for a few minutes, especially when trying to interrupt. Ironically it seems to happen only at slower Bluetooth update rates. I have a light sensor and an ADS1115 on the 5V pins which might be causing a power draw issue but this was no issue with BLE. 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include "BluetoothSerial.h"

#include <Wire.h>
#include <Adafruit_ADS1015.h>
//#include <ArduinoJson.h>
 
BluetoothSerial SerialBT;

int count = 0;
bool sensorEnabled = false;
bool adcEnabled = false;
bool usbSerialEnabled = false;

char received;
//char ratioBuf[6], adcBuf[6];

// HEG VARIABLES
bool DEBUG = false;
bool VIEW_ADC_VALUE = false;
bool SEND_DUMMY_VALUE = false;

// PUT IR IN 13, RED IN 12
const int IR = 13;
const int RED = 12;
const int LED = 5; // Lolin32V1.0.0 LED on Pin 5
int16_t adc0; // Resulting 16 bit integer

//Setup ADS1115
Adafruit_ADS1115 ads;

float Voltage = 0.0;
float range = 32767; // 16 bit ADC (15 bits of range minus one)
float gain = 0.256; // +/- V
float bits2mv = gain / range;

//Signal flags
bool first_led = false; // Bool to alternate LEDS
bool badSignal = false;
bool signalDetermined = false;

//Counters
int ticks0, ticks1, ticks2, ticks3, ticks4 = 0;

//Scoring variables
long redValue = 0;
long irValue = 0;
float redavg = 0;
float iravg = 0;
float ratio = 0;
float baseline = 0;

float score = 0;

[... 820 lines cut ...]
Backtrace: 0x40085841:0x3ffc05b0 0x400892b9:0x3ffc05d0 0x4008994b:0x3ffc05f0 0x40081f45:0x3ffc0610 0x40089aab:0x00000000

Core 1 register dump:
PC      : 0x4017e1ce  PS      : 0x00060f34  A0      : 0x8008e46a  A1      : 0x3ffd1de0  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd1910  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc829c  A9      : 0x3ffc8260  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x8008c8c8  A13     : 0x3ffd1ce0  
A14     : 0x00000000  A15     : 0x3ffd1aa0  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4017e1ce:0x3ffd1de0 0x4008e467:0x3ffd1e00

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
92620

moothyknight, 2019-01-22:
Another stack trace:
0x40085841: r_assert_param at ?? line ?
0x40085779: r_assert_param at ?? line ?
0x40085780: r_assert_param at ?? line ?
0x40089aae: btdm_bb_isr at intc.c line ?
0x40085841: r_assert_param at ?? line ?
0x40085841: r_assert_param at ?? line ?
0x400892b9: r_rwbt_isr at ?? line ?
0x4008994b: r_rwbtdm_isr_wrapper at intc.c line ?
0x40081f45: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/xtensa_vectors.S line 1105
0x40089aab: btdm_bb_isr at intc.c line ?
0x4017e1fa: esp_vApplicationWaitiHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 66
0x4017e1fa: esp_vApplicationWaitiHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 66
0x4008e467: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3564

copercini, 2019-01-22:
`Interrupt wdt timeout on CPU0` means that you should feed the WDT timer on loops, calling `feedLoopWDT();` or `delay(10);` with 10 or more

moothyknight, 2019-01-22:
So should I change the 1ms delay to 10 or change it to feedLoopWDT(). I have some processes that are running at 2ms intervals so I also need to desynchronize then.

moothyknight, 2019-01-22:
Okay, fix: changed the delay after the BTserial send to 10ms, and I have a 1ms delay in the main loop for good measure, now my heap at slower response rates is restored

closed 2019-01-22 by copercini