espressif/arduino-esp32 #2368: ESP32 not working

psimec, 2019-01-22:
My esp32 stopped working today... I was messing around with some code for soil moisture sensor and when I tried upload code I got an error:"a fatal error occurred failed to connect to esp32 timed out waiting for packet header". I have no red light on my esp32 like I had before. Can someone help me? Is my ESP dead and I needed to buy a new one?

ArTyGill, 2019-01-23:
@psimec Measure the power supply at pin 3.3V

ArTyGill, 2019-01-23:
The voltage regulator may have failed.

remcoNL, 2019-01-30:
This worked fo me:

> You can try holding down the button marked "BOOT" or "FLASH" while esptool.py is "Connecting...". If you see esptool.py connect then you can release this button.

https://github.com/espressif/arduino-esp32/issues/2368

closed 2019-08-15 by stale[bot]