espressif/arduino-esp32 #2370: ESP32 no connect to a wifi network

03david7069, 2019-01-23:
### Hardware:
Board:							ESP322-WROOM-32
Core Installation/update date:			I don´t know but i bought the module a week ago
IDE name:							Arduino IDE -- DOIT ESP32 DEVKIT V1
Flash Frequency:					80Mhz
PSRAM enabled:                                          ?no?
Upload Speed:						921600
Computer OS:                                            Windows 7

### Description:
I bought two ESP32 a week ago, and i have been trying to connect to a my wifi network but both modules not connect and throw me the same error.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFi.h>

const char* ssid     = "FAMILIA ANGARITA";
const char* password = "T63530193";

IPAddress local_IP(192, 168, 0, 115);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);

void setup() {
  Serial.begin(115200);
  if (!WiFi.config(local_IP, gateway, subnet)) {
    Serial.println("Error al configurar");
  }
  Serial.print("Conectandose a: ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi conectado!");
  Serial.print("IP asignada: ");
  Serial.println(WiFi.localIP());
  Serial.print("MAC: ");
  Serial.println(WiFi.macAddress());
  Serial.print("Mascara de subred: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("IP de la puerta de enlace: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("DNS: ");
  Serial.println(WiFi.dnsIP());

}

void loop() {

}
```
I used this info from mi computer that is connected to a my network:

[image]

### Debug Messages:
```
Conectandose a: FAMILIA ANGARITA
....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
.....[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND

```

Thx and sorry for any mistake, i am learning english :D

Jeroen88, 2019-01-23:
What if you skip this:
```
  if (!WiFi.config(local_IP, gateway, subnet)) {
    Serial.println("Error al configurar");
  }

```

Your SSID contains a space, might that be the problem?

Tip: never publish your WiFi credentials on the internet...!

baaridunnasr, 2019-01-23:
The Code I'm using is very similar but I get a 

"AUTH_EXPIRE" Failure while connecting to a Mobile Hotspot. Haven't tried connecting to a normal WiFi yet.

03david7069, 2019-01-25:
> What if you skip this:
> 
> ```
>   if (!WiFi.config(local_IP, gateway, subnet)) {
>     Serial.println("Error al configurar");
>   }
> ```
> Your SSID contains a space, might that be the problem?
> 
> Tip: never publish your WiFi credentials on the internet...!

Hi, sorry for no test tis before but my school no let me free time.

I changed my SSID for "ANGARITA", but the problem continues, my router is a technicolor TC8305C, you know if exits the possibility that the router adds a no visible character in the SSID, i think this because i have a smartTV, and when i try to connect using the option that you can ingress the SSID, the TV no connect to gateway 
[image]

But if you select the SSID from a list, the TV is connected succesfully.

Thx for all participations :D

baaridunnasr, 2019-01-26:
Hi, @03david7069 

Can you try without 

```
IPAddress local_IP(192, 168, 0, 115);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);
```

and let us know? Let the router assign the local IP and see what it prints out using

```
Serial.print("local IP : ");
Serial.println(String(WiFi.localIP()));
Serial.print("SSID : ");
Serial.println(String(WiFi.SSID()));
Serial.print("MAC : ");
Serial.println(String(WiFi.macAddress()));
```

If the connection still fails, then attempt the WiFiScan example in Examples to see if your ESP32 can see the WiFi network.

Personally, this is the code I defined in another header file to use for connecting and disconnecting to WiFi. You can adapt it as you see fit.

```
//function details
//name : wifiConnect
//purpose : connects to given ssid using password for given number of tries. updates the wifiStatus.
//argument(s) : char *SSID, char *PASS, unsigned short tries
//returns : void
void ESP32Wifi::wifiConnect(char *SSID, char *PASS, unsigned short tries)
{
    _SSID = SSID;
    _PASS = PASS;
    wifiStatus = WiFi.status() == WL_CONNECTED;
    Serial.println(String(tries));
    Serial.println(String(wifiStatus));
    while ((tries > 0) && (!wifiStatus))
    {
        wifiClear();
        WiFi.begin(_SSID, _PASS);
        delay(5000);
        wifiStatus = WiFi.status() == WL_CONNECTED;
        Serial.println(String(tries));
        Serial.println(String(wifiStatus));
        tries--;
    }
}
```

and

```
//function details
//name : wifiDisconnect
//purpose : disconnects from the network and switches off the radio. sets wifiStatus to false.
//argument(s) : void
//returns : void
void ESP32Wifi::wifiDisconnect()
{
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    wifiStatus = false;
}
```

03david7069, 2019-01-27:
Hi, i finally find the problem, previously i comment that my TV did not find the network if i wrote the SSID and did not choose a from the APs list, then i thought that the SSID had non-visible characters, for test this i edited the example of scanning networks to shows me the ASCII code of all characters of SSID:
```
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            
            String net = WiFi.SSID(i);            
            for (int x = 0; x < net.length(); ++x){
              Serial.print((int)net[x]);
              Serial.print(" ");
            }
            Serial.println("\n----------");
 
            delay(10);  
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
And this is the output:
```
Setup done
scan start
scan done
4 networks found
1: ANGARITA   (-47)*
65 78 71 65 82 73 84 65 32 32 
----------
...
```
The SSID have 2 space characters (decimal 32 on ASCII) at end of SSID, only i need add two space characters at SSID when i will connect, and now works perfectly.

NOTE: I checked why those 2 space characters on my router config, but i don´t found nothing.

Thx for all participations :D

closed 2019-01-27 by 03david7069

sjames1958gm, 2019-07-22:
I am seeing a similar issue, sometimes the Wifi will not connect even after several resets. And then it connects almost everytime. This is really unreliable, the Wifi code is running in the second core.

```
void wifiSetup()
{
  xTaskCreatePinnedToCore(
      wifiTask, /* Function to implement the task */
      "Wifi Task", /* Name of the task */
      10000,  /* Stack size in words */
      NULL,  /* Task input parameter */
      2,  /* Priority of the task */
      &wifiTaskHandle,  /* Task handle. */
      0); /* Core where the task should run */
}
```

```
void wifiSetupInternal() {

  gWifiStatus = 1;
  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  esp_wifi_set_ps(WIFI_PS_NONE);
  Serial.print("MAC: ");
  Serial.println(WiFi.macAddress());
  gmacAddr = WiFi.macAddress();
  // commented out to allow WiFi to pick address
  //  WiFi.config(ip, gateway, netmask);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
```

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
34

Version: esp32 - version 1 2019-05-02 - test D

Connecting to TXTdev
MAC: 24:0A:C4:1D:5E:74
.........................
```

tim-jh, 2020-04-30:
Adding two spaces after the end of the SSID worked for me :)

messplay, 2020-08-27:
Works for me, two spaces at the end of the SSID , modem arrys tg862. Thanks

ugifractal, 2020-11-01:
The SSID also seem case sensitive, I  fixed my issue by using same character case

JoePortilla, 2021-03-07:
Works for me too, I have one space at the end of the SSID. Thanks!