espressif/arduino-esp32 #2375: xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory

1TekPro, 2019-01-24:
This issue has been brought up many times already but I am still having issues with this. 
I am using a Heltec Wifi_Kit_32 board and a blank sketch on windows 10 - 64bit. I have tried multiple versions of arduino IDE (1.6.13 - 1.8.8).
I have followed the steps in all of the of topics solutions including https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/InstallGuide/windows.md
The only step that doesnt appear to do anything (other than a slight delay, then prompt returns) is; "git submodule update --init --recursive" in the bash session. Is this supposed to do anything visual?
None the less, I have uninstalled, deleted and retried 4 times now and get the same error. Am I missing something?

1TekPro, 2019-01-24:
I solved my issue. For some reason I had to run the IDE with administrative privileges.

1TekPro, 2019-01-25:
Just an update, when I follow these steps, it generates the same issue with WiFi_Kit_8 after following these steps. If I delete the 32 files, kit_8 works again.

1TekPro, 2019-01-27:
Ok so I think I figured most of this out incase anyone else might have the same issue.
It seams that access was restricted to the working directory of "appdata" so I changed the directory in the preferences.txt file to the arduino folder in Program Files(x86).
This fixed the Heltec WiFi_Kit_32 but not for WiFi_Kit_8. For series 8 boards, the nodemcu 1.0(esp-12E module) board does work.

closed 2019-01-27 by 1TekPro

Tranks27, 2022-11-22:
Sometimes I get this error too even if the compilation has been working a few days ago. In such case, my go-to solution is to just delete the .platformio folder found here C:\Users\xxx\.platformio. Then reopen the VScode and platformio will be automatically reinstalled. Works like a charm. :)