espressif/arduino-esp32 #2376: BLECharacteristic Requires Two Calls for Value

czuvich, 2019-01-24:
I'm currently targeting the 1.0.1 Arduino core using an ESP32 Feather HUZZAH.  The ESP32 is acting a server and I have a few custom services with characteristics.

I noticed that when I upgraded from 1.0.1, my client (also an ESP32) was not getting the  BLERemoteCharacteristic values from my services unless I called the getValue twice.  So I dug further, and I noticed that unless I set the value in the BLECharacteristic before advertising, the value does not actually get set until the second callback.  So here's what I'm seeing with execution flow:

Scenario 1 - Missing Value on first Client.getValue()
1) Server - Setup the characteristic, but do not assign a value yet.
2) Start advertising.
3) Once client connects and asks for the value, it returns empty.
4) A second call will return the correct value.

Scenario 2 - Has Value on first Client.getValue()
1) Server - Setup the characteristic and assign the value.
2) Start advertising
3) Once client connects and asks for the value, it returns properly.

In both scenarios, I see the callback being invoked on the server when a characteristic read.  Is this by design?  I do not see this behavior in 1.0.0.

czuvich, 2019-01-24:
I double-checked my debug logs, and I am seeing the value being set, but it's not showing up until a second call (for any characteristic).  I rolled back to 1.0.0, and I don't see this behavior.  I'm seeing it on both my ESP32 client and iPhone.  Any workarounds?  I tried calling BLEValue.commit() for grins, but that didn't work either.

```
17:07:10.005 -> [D][BLECharacteristic.cpp:631] setValue(): >> setValue: length=4, data=342e3133, characteristic UUID=9006c842-0595-4d5a-a317-69c01c661d77
17:07:10.005 -> [D][BLECharacteristic.cpp:638] setValue(): << setValue
```

czuvich, 2019-01-24:
Ok.. I found the issue in BLECharacteristic.  It's in line 402.  The callback needs to be invoked BEFORE committing it to the response in line 387.

everslick, 2019-01-25:
ping @chegewara

chegewara, 2019-01-25:
Hi,
there is no need to ping me. I will check it and respond when i have time.

czuvich, 2019-02-05:
@chegewara No, but I don't think I see anything that will address this issue.  I looked here
https://github.com/nkolban/ESP32_BLE_Arduino/pulls?q=is%3Apr+is%3Aclosed

Is there somewhere else I should be checking?

chegewara, 2019-02-05:
There is few bugfixes in master on this repository. I have to make few more fixes before i push to arduino library.

czuvich, 2019-02-05:
Ok.  I'll wait until it gets pushed.  I have some workarounds for now.  Thanks for the update!

czuvich, 2019-02-16:
I went ahead and grabbed the updates from https://github.com/chegewara/esp32-snippets/tree/master/cpp_utils and everything works great!  There were a few other issues I've noticed, and it looks like those have been fixed, too.  Thanks chegewara!

closed 2019-08-15 by stale[bot]