espressif/arduino-esp32 #2381: Can't upload file with PC, but can view serial monitor via smartphone

StijnVerhelst, 2019-01-26:
Dear,
I've a problem with the ESP32. 
When I try to upload a file, it won't work (but it is succesfully connected to the PC). When I try to view some serial monitor, I also don't get a response.

When I try to view the serial monitor via my smartphone, everything is working just fine.

Is there any way we can upload a compiled file to the ESP32 using a smartphone? Or has anyone some ideas to solve this problem?

Kind regards,
Stijn Verhelst

StijnVerhelst, 2019-01-27:
Dear,

> When you say "successfully connected to the pc", you mean the correct drivers are installed AND you have the correct com port selected?

Yes, indeed, the correct drivers are succesfully installed, and I've selected the correct COM-port. Here's an image that shows the device: https://www.mupload.nl/img/f5oitmh8zmj.png

> Code can be uploaded by a smartphone using ArduinoDroid and ArduinoCommander, but that would not solve your issue.

I've also seen this app, but does it also work with ESP32?

> What serial speed (baud rate) do you have set? Do you get anything from the serial monitor at all? What happens when you reset the esp32 with the monitor open?

The baud rate is 115200 baud. I didn't get anything from the serial monitor at all.
When I reset the ESP32 with the serial monitor open, I also get nothing.

When I connect the ESP32 with my phone using this app (https://goo.gl/XZomgm), I can succesfully connect with the ESP32, I get serial data (using baud rate 115200).

I also have another ESP32 at home, and when I plug this device in into my PC, everything is working fine. I have serial data and so on.

Is it possible that this problem is caused by a uploaded sketch or something else?

Kind regards,
Stijn Verhelst

1TekPro, 2019-01-27:
With the error shown in doy9gb1.png, it would appear that there is one of two possible issues. 
Baud rate mismatch:
Go to device manager, open the properties window of the com port used for the esp32 and double check the baud rate isn't set to something else on the driver side.
Esp32 not entering flash mode:
Hold the PRG down while pressing RST, then try to upload a blank sketch. Sometimes (although rare) you must Ground D0 during the entire upload process.

1TekPro, 2019-01-27:
"Is it possible that this problem is caused by a uploaded sketch or something else?"

I have seen sketches freeze and temporarily "brick" devices, but this is unlikely in your case as you get serial data from it on your phone.

StijnVerhelst, 2019-01-27:
@1TekPro 
After double-checking everything you said, the ESP32 works now just fine!
Using this guide: http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/ I succesfully sended a blank sketch.

Thanks a lot, you saved my month!!

closed 2019-01-27 by StijnVerhelst