espressif/arduino-esp32 #2384: ESP32 with FONA library

CaNiRaspUsomethinG, 2019-01-27:
Trying to connect FONA library to one of my ESP32 hardware serials. What changes do I need to do?

This has not helped me:
https://github.com/espressif/arduino-esp32/issues/426#issuecomment-307115386

It causes this compilation error:
arduino.ar(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'
sketch/FONAtest.ino.cpp.o:(.bss.Serial2+0x0): first defined here
collect2: error: ld returned 1 exit status

In my scripts I usually use this without problems to initialize my serial port:
#include <HardwareSerial.h> 
HardwareSerial Ser1_33_32_GSM(1);

void setup() {
Ser1_33_32_GSM.begin(9600,SERIAL_8N1,33, 32); 
....
}

My board is a ESP32 Wroom DOIT Devkit  // My ESP32 rep. is still 1.0.0.

Thank you!

Jeroen88, 2019-01-27:
This place is for tracking issues with the ESP32 core, not a helpdesk!

Your error has nothing to do with FONA directly, you define Serial2 for a second time.

If you want to use Serial2 you should uncomment this line:
https://github.com/adafruit/Adafruit_FONA/blob/bf59657b42e4fd59b93434925413a1b16afc23a7/examples/FONAtest/FONAtest.ino#L45
and change &Serial1 into &Serial2.  

It took me some time to find a working combination of pins on my ESP32 Heltec and NodeMCU, so maybe you have to try several pins after you have solved the compilation error.

CaNiRaspUsomethinG, 2019-01-27:
Sorry for posting in the wrong place...I will mind that for my future posts. @Jeroen88 : In need to use Serial1 exactly on ports 33,32. Generally this works well, if I send AT+T commands directly to the GSM using this lines of code:
#include <HardwareSerial.h>
HardwareSerial Ser1_33_32_GSM(1);
void setup() {
Ser1_33_32_GSM.begin(9600,SERIAL_8N1,33, 32);
Ser1_33_32_GSM.println("AT");
}
Just uncommenting the lines (&Serial1) in Fona test example doesn't work. I don't know how to implement the lines above in the Fona example sketch or (if necessary) in the Adafruit_FONA.cpp.

Jeroen88, 2019-01-27:
Take the FONAtest.ino example from https://github.com/adafruit/Adafruit_FONA/blob/master/examples/FONAtest/FONAtest.ino

If you want to use Serial1 uncomment the line:
`
HardwareSerial *fonaSerial = &Serial1;
`
and leave out the SoftwareSerial.

In
https://github.com/adafruit/Adafruit_FONA/blob/bf59657b42e4fd59b93434925413a1b16afc23a7/examples/FONAtest/FONAtest.ino#L63

change to:
`
fonaSerial->begin(9600, 33, 32);
`

CaNiRaspUsomethinG, 2019-01-29:
Works like a charm! Thank you !

closed 2019-01-30 by CaNiRaspUsomethinG