espressif/arduino-esp32 #2385: UPDATE example takes up a lot of ESP32 memory

DeltaVetal26, 2019-01-27:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		v1.0.1 / Jan / 2019
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                          Windows 10

### Description:
Describe your problem here
Hi!

I want to implement the function of remote firmware update ESP32 via http. For this, I found a library of updates. First, I just wanted to download the AWS_S3_OTA_Update.ino example, where to add the usual Blink example.
When compiling, I noticed that the sketch occupies 51% of the memory, which is very strange. As far as I understand, for OTA to work, it is necessary to use no more than 50% of the memory so that you can load a new firmware into the chip.
The question is, why so much memory uses this sketch? I can not use external memory, because I need to wait a long time, but there is no time. Thank!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/**
   AWS S3 OTA Update
   Date: 14th June 2017
   Author: Arvind Ravulavaru <https://github.com/arvindr21>
   Purpose: Perform an OTA update from a bin located in Amazon S3 (HTTP Only)
   Upload:
   Step 1 : Download the sample bin file from the examples folder
   Step 2 : Upload it to your Amazon S3 account, in a bucket of your choice
   Step 3 : Once uploaded, inside S3, select the bin file >> More (button on top of the file list) >> Make Public
   Step 4 : You S3 URL => http://bucket-name.s3.ap-south-1.amazonaws.com/sketch-name.ino.bin
   Step 5 : Build the above URL and fire it either in your browser or curl it `curl -I -v http://bucket-name.ap-south-1.amazonaws.com/sketch-name.ino.bin` to validate the same
   Step 6:  Plug in your SSID, Password, S3 Host and Bin file below
   Build & upload
   Step 1 : Menu > Sketch > Export Compiled Library. The bin file will be saved in the sketch folder (Menu > Sketch > Show Sketch folder)
   Step 2 : Upload bin to S3 and continue the above process
   // Check the bottom of this sketch for sample serial monitor log, during and after successful OTA Update
*/

#include <WiFi.h>
#include <Update.h>

WiFiClient client;

// Variables to validate
// response from S3
int contentLength = 0;
bool isValidContentType = false;

// Your SSID and PSWD that the chip needs
// to connect to
const char* SSID = "SSID";
const char* PSWD = "pass";

// S3 Bucket Config
String host = "mywebsite.com"; // Host => bucket-name.s3.region.amazonaws.com
int port = 80; // Non https. For HTTPS 443. As of today, HTTPS doesn't work.
String bin = "/blink.ino.bin"; // bin file name with a slash in front.
[... 161 lines cut ...]
  // Wait for connection to establish
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print("."); // Keep the serial monitor lit!
    delay(500);
  }

  // Connection Succeed
  Serial.println("");
  Serial.println("Connected to " + String(SSID));

  // Execute OTA Update
  execOTA();
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(5000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(5000);
  // chill
}
```

### Debug Messages:
I cannot receive the debug message, because the ESP refuses to work from the PC port due to poor power supply.

chegewara, 2019-01-27:
You can occupy up to 100% memory, because its not whole flash size considered, but partitions size. If you have selected proper partition scheme selected then you have 2 partitions and your app occupy only 51% from one partition.

DeltaVetal26, 2019-01-27:
@chegewara 

I am using the default partition scheme. So everything is fine, Many thanks for the help!

1TekPro, 2019-01-27:
If you are satisfied with the solution/answer, please close the issue.  :)