espressif/arduino-esp32 #2387: Can you acces the serial data through pins?

StijnVerhelst, 2019-01-27:
Dear,
For my project I need to access the pins that are connected to the UART to USB chip (in my case the CP2102).
At the moment I need to connect the ESP32 with my PC using the micro-usb port on the ESP32. For making everything clean I want to plug everything in into a self-made PCB (jlcpcb.com actually) and connect this to a USB type A. For this I need to access the ports that I otherwise have acces to when using the micro-USB on the ESP32.

**In short:** Can I upload a sketch/view serial data using pins on the ESP32 board?

Kind regards,
Stijn Verhelst

Petrovjan, 2019-01-27:
I suppose this will sound a bit obvious, but you best bet is to access the RX/TX pins and connect them to your PC through a standalone USB-TTL converter you can get for 1$.

StijnVerhelst, 2019-01-27:
@Petrovjan Thanks for your reply!
Altough this might be the most "safe" solution, do you think it's possible to solder two wires to D- and D+ from the micro-USB of the ESP32 to access the serial data?

Kind regards,
Stijn Verhelst

Petrovjan, 2019-01-27:
Well, sure, but in that case I'd rather stick a micro usb cable into the ESP32 board, but the cable after a few cm and solder its wires to the USB A connector that you need. You can't really break anything if you do it like that, while trying to solder the wires directly to the tiny micro usb pads sounds like a good way to damage the whole board...

1TekPro, 2019-01-27:
I recommend using an adaptive cable for this.
Examples:
[https://www.amazon.com/dp/B01M098EAG?aaxitk=H1uHfL8LqakVQbUmocMyjQ&pd_rd_i=B01M098EAG&pf_rd_p=3ff6092e-8451-438b-8278-7e94064b4d42&hsa_cr_id=8752329850501&sb-ci-n=asinImage&sb-ci-v=https%3A%2F%2Fimages-na.ssl-images-amazon.com%2Fimages%2FI%2F61XJiIlcdtL.jpg&sb-ci-a=B01M098EAG](url)
[https://www.amazon.com/AmazonBasics-Male-Micro-Cable-Black/dp/B0711PVX6Z/ref=asc_df_B0711PVX6Z/?tag=hyprod-20&linkCode=df0&hvadid=198093101467&hvpos=1o1&hvnetw=g&hvrand=15041796834197631517&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9012266&hvtargid=pla-359221356227&psc=1](url)

closed 2019-08-15 by stale[bot]