espressif/arduino-esp32 #2388: ESP32 Platform unknown on Arduino IDE

malebuffy, 2019-01-27:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?

Upload a sketch to an ESP32 with Arduino IDE

Describe your system( Hardware, computer, O/S, core version, environment).

Windows 10 on MAC 

Describe what is failing

When uploading a sketch I get:

Board esp32 (platform esp32, package esp32) is unknown
Error compiling for board ESP32 Dev Module.

I have installed the ESP32 though the Board Manager like in the Espressif tutorial

.
Show the shortest possible code that will duplicate the error.

I am using the default/blank code:

Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

joeybab3, 2019-01-28:
I also ended up dealing with this issue for a week or so, I ended up deleting everything relating to ESP32 from the appdata/roaming/arduino15(or wherever its stored for you) ass well as uninstalling the board and then reinstalling and it worked after that for me.

1TekPro, 2019-01-28:
Speaking of "appdata" check out my post if joeybab3's solution doesn't work for your issue.

https://github.com/espressif/arduino-esp32/issues/2375

malebuffy, 2019-01-28:
I got it working by deleting everything and reinstalling manually from the zip. Thanks guys.

1TekPro, 2019-01-28:
@malebuffy  If you are satisfied with the solution, please close this issue. Thank you.  :)

closed 2019-01-28 by malebuffy

JaspaJami, 2019-02-27:
> Hope it helps someone :)
Thanks @stankusaudrius i had exactly same problem

DevSpur, 2019-03-06:
Hi everyone I had the same error. All I had to do was locate to the download location and delete old versions eg: I had installed V1.0.2 and had to delete V1.0.0 and V1.0.1 The location is:  ⁨Library⁩ ▸ ⁨Arduino15⁩ ▸ ⁨packages⁩ ▸ ⁨esp32⁩  for anyone that does not know

rinelb, 2019-04-01:
Got same error , uninstalled and reinstalled fixed it

robynjayqueerie, 2019-04-25:
Deleting the V1.0.0 library from ▸ ⁨packages⁩ ▸ ⁨esp32⁩ worked for me

SteveRMann, 2019-05-18:
> I got the same error after updating ESP32 board package from 1.0.0 to 1.0.1.
> 
> I then noticed that IDE was for some reason looking for boards.txt file in 1.0.0 library folder, which was almost empty but not deleted. I deleted the whole folder for 1.0.0 and error disappeared.
> 
> Mine was in C:\Users\Username\AppData\Local\Arduino15\packages\esp32\hardware\esp32\
> 
> Hope it helps someone :)

It did! THANK YOU.

I was looking for the solution for days when I stumbled on your explanation.  Most perplexing was that I installed the IDE and ESP32 libraries and boards on a different PC and everything compiled without error.   But on my main PC, it wouldn't.  When I looked in
`C:\Users\steve\AppData\Local\Arduino15\packages\esp32\hardware\esp32`
I found a prior version of the ESP32 boards.  I deleted the old version folder and now the sketches compile without error.

Scimac, 2019-12-14:
@stankusaudrius Thanks for the help. It was indeed looking for the wrong packages.

You can also check the settings path in preference.txt and change only when android ide is not working

PantalaLabs, 2022-03-28:
go back to arduino IDE and reinstall 
https://dl.espressif.com/dl/package_esp32_index.json

work fine here !

me-no-dev, 2022-03-28:
@PantalaLabs this is a very old link to the JSON. We have not updated that in ages. Please see our documentation for the new link and use that.

SteveRMann, 2022-03-28:
**"Please see our documentation for the new link and use that."**
And, that link is, _where_?

me-no-dev, 2022-03-29:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html