espressif/arduino-esp32 #2391: Wrong "SPIFFS File system is full"?

Menion2k, 2019-01-28:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE 1.8.7
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
mkspiffs 0.2.3 seems to report file system full even if it should not

I have a filesystem, flat, no directory, of size of 1249718 bytes (1306624 occupied on NTFS file system), composed by 27 files.
When trying to pack it with the command (that match the arduino default partition type);

```
mkspiffs.exe -c "c:\Users\silveria\Documents\sloeber-workspace\dimmer\data" -p 256 -b 4096 -s 1503232 data.bin
/bootstrap.min.css
/bootstrap.min.js
/Chart.bundle.js
/feather.min.js
/font1.woff2
/font10.woff2
/font11.woff2
/font12.woff2
/font13.woff2
/font14.woff2
/font15.woff2
/font2.woff2
/font3.woff2
/font4.woff2
/font5.woff2
/font6.woff2
/font7.woff2
/font8.woff2
/font9.woff2
/index.html
/jquery-3.3.1.min.js
SPIFFS_write error(-10001): File system is full.

error adding file!
```

1503232 is the SPIFFS reserved size, which is greater than the filesystem I am trying to create. Apparently somewhere after 1080KB the tool report File system is full. Moreover the SPIFFS is then compressed, so in the end more data can be stored
Bye

1TekPro, 2019-01-28:
I'm not sure from the wording at the end of your post, if you resolved your issue or not. But...
Select a larger SPIFFS size under Tools > Flash Size, or delete some files. 
Even if your computer says that the files are smaller than the selected SPIFFS size, you can still get this error: this has to do with block sizes, and metadata like file and folder names that take up space as well.

Menion2k, 2019-01-28:
I removed some files to check when I manage to create the SPIFFS.
Indeed trying to increase blocksize improve the situation till 16KB of block size when I manage to create the SPIFFS with all the files. I know that it is not ok, since the blocksize shall reflect the NOR block size and I am a bit surprised by how much overhead the SPIFFS adds to the actual data. I mean, here I get ~400KB of overhead in a 1MB of data, which is an insane big value IMHO

1TekPro, 2019-01-28:
You could also decrease the block size to reduce wasted space. It may be worth checking the average file size and configuring the block size close to that number.

Menion2k, 2019-01-28:
Decrease block size makes the things worse

1TekPro, 2019-01-28:
If your sketch doesn't take much space, you can increase the spiff size.

closed 2019-08-15 by stale[bot]