espressif/arduino-esp32 #2394: WifiClient & WiFiClientSecure relaibility & disconnections when being used by PubSubClient

macedolfm, 2019-01-28:
### Hardware:
Board:							Heltec_WiFi_LoRa_32
Core Installation version:			1.0.0 & 1.0.1
IDE name:						VS2017 + VisualGDB
Flash Frequency:					80Mhz
PSRAM enabled:                                    
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:

I am experiencing random fault results when using PubSubClient + (WiFiClient or WiFiClientSecure). For what I have tested these Clients are very unstable: gets disconnected during publish for no reason, may get stuck connecting for minutes, crashes the ESP32 sending long (over 1024 bytes), get stuck if MQTT server stops, get stuck if WiFi stops and all sort of unexpeceted problems all in a very random fashion.

I´ve been trying everything for the last couple days and letting the MCU run overnight to collect error data and just can´t figure out what is wrong (I am NOT a inner guts expert), tried with "arduino-esp32-master" version 1.0.0 & 1.0.1, even tried [this guy tips](http://www.iotsharing.com/2017/08/how-to-use-esp32-mqtts-with-mqtts-mosquitto-broker-tls-ssl.html?m=1) and a ["this partial release"](https://github.com/espressif/arduino-esp32/tree/a38ffe58fe1d73a56295edb0c6051a46932e0175) of WiFiClientSecure but still, lots of trouble.

I managed to addapt the own library example to show case the many problems on a MQTT permanent connection sketch:

```
#include <WiFi.h> 
#include <WiFiClientSecure.h>

#define  ARDUINOJSON_USE_LONG_LONG  1
#define  ARDUINOJSON_USE_DOUBLE     1
#include <ArduinoJson.h>

#define MQTT_MAX_PACKET_SIZE 4096
#include <PubSubClient.h>

#include <SerialDebug.h>

// Lets Encrypt root CA-CRT

const char* test_root_ca = "-----BEGIN CERTIFICATE-----\n" \
  "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
  "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
  "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
  "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
  "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
  "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
  "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
  "SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
  "Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
  "a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
  "/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
  "AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
  "CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
  "bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
  "c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
  "VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
  "ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
  "MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
  "Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
  "AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
  "uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
  "wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
  "X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
  "PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \
  "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
  "-----END CERTIFICATE-----";

const char* foo_4000 = "0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789";

WiFiClientSecure wifi_client;
PubSubClient mqtt_client(wifi_client);

const char* wifi_ssid     = "********";
const char* wifi_password = "********";

const char* server   = "gw1.ivigilant.com.br";
const int   port     = 8883;
[... 147 lines cut ...]
(E p:^50006 mqttConnect C1) MQTT_CONNECTING
(E p:^12120 mqttConnect C1) MQTT_CONNECT_OK
(E p:^160018 mqttConnect C1) MQTT_CONNECTING
(E p:^12623 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3353289 loop C1) MQTT_PUB_ERR
(E p:^0000 mqttConnect C1) WIFI_CONNECTING
......
(E p:^6059 mqttConnect C1) WIFI_CONNECT_OK
(E p:^3942 mqttConnect C1) MQTT_CONNECTING
(E p:^11753 mqttConnect C1) MQTT_CONNECT_OK
(E p:^151967 mqttConnect C1) MQTT_CONNECTING
(E p:^14284 mqttConnect C1) MQTT_CONNECT_OK
(E p:^40006 mqttConnect C1) MQTT_CONNECTING
(E p:^25991 mqttConnect C1) MQTT_CONNECT_OK
(E p:^122003 mqttConnect C1) MQTT_CONNECTING
(E p:^11105 mqttConnect C1) MQTT_CONNECT_OK
(E p:^90011 mqttConnect C1) MQTT_CONNECTING
(E p:^14638 mqttConnect C1) MQTT_CONNECT_OK
```

And just running it for a few min utes you can see that the MQTT has to reconnect over & over to get the job done.

Crying for HELP here, as I am in a tight corner here.

Cheers evryone !

1TekPro, 2019-01-28:
if (mqtt_client.connected()) 
  {
    mqtt_client.disconnect();
  }

macedolfm, 2019-01-29:
> @macedolfm I would take a look at where your error log says "MQTT_CONNECT_ERR". That debug is from an if statement in "void mqttConnect()". It contains:
> If (mqtt_client.connected())
> {
> mqtt_client.disconnect();
> }

Well, this was just an overkill, as mqttConnect() is only called when mqtt_client.connected() == false in the main loop() and during setup() when everyone is disconnected(). The reason behind this assurance is that PubSub.connected() && PubSub.disconnect() makes a deep clean in the WiFiClient as in:

```
boolean PubSubClient::connected() {
    boolean rc;
    if (_client == NULL ) {
        rc = false;
    } else {
        rc = (int)_client->connected();
        if (!rc) {
            if (this->_state == MQTT_CONNECTED) {
                this->_state = MQTT_CONNECTION_LOST;
                _client->flush();
                _client->stop();
            }
        }
    }
    return rc;
}

and....

void PubSubClient::disconnect() {
    buffer[0] = MQTTDISCONNECT;
    buffer[1] = 0;
    _client->write(buffer,2);
    _state = MQTT_DISCONNECTED;
    _client->flush();
    _client->stop();
    lastInActivity = lastOutActivity = millis();
}
```

Will try without it anyway...thx

1TekPro, 2019-01-29:
In the main loop you call it.
If there is no error, then mqqconnect. Then mqq disconnect.

macedolfm, 2019-01-29:
> In the main loop you call it.
> If there is no error, then mqqconnect. Then mqq disconnect.
> 
> At first glance, It looks like this might be backwards.
> if (mqtt_client.publish("pub", mqtt_buff, false))
> {
> //debugE("MQTT_PUB_OK");
> }
> else {
> debugE("MQTT_PUB_ERR");
> 
> Change this to true and uncomment the debug?

As per the PubSub lib source code :

```
boolean PubSubClient::publish(const char* topic, const uint8_t* payload, unsigned int plength, boolean retained) {
    if (connected()) {
        if (MQTT_MAX_PACKET_SIZE < MQTT_MAX_HEADER_SIZE + 2+strlen(topic) + plength) {
	        // Too long
			Serial.println("MQTT_MAX_PACKET_SIZE");
            return false;
        }
        // Leave room in the buffer for header and variable length field
        uint16_t length = MQTT_MAX_HEADER_SIZE;
        length = writeString(topic,buffer,length);
        uint16_t i;
        for (i=0;i<plength;i++) {
            buffer[length++] = payload[i];
        }
        uint8_t header = MQTTPUBLISH;
        if (retained) {
            header |= 1;
        }
        return write(header,buffer,length-MQTT_MAX_HEADER_SIZE);
    }
	Serial.println("MQTT NOT CONNECTED");
    return false;
}
```

The function returns the number of bytes published, so it returns "0" == FALSE, meaning the that te Publishing as failed, but anyway the issua is not here, the fact is that all the underlaying libraries (PubSub, WiFiClient WiFiClientSecure, all the way don to start_ssl_client(), lwip_socket() and lwip_connect() don´t like weeak WiFi spots, large payload, nor slow connections. I am testing a new promissing multitask version wich is behaving muche better, but not stelar, will let it running overnight and check the results. Will be back, thanks.

1TekPro, 2019-01-29:
@macedolfm  Are the ( ) and [ ] supposed to be different?
      msg["free_mem"] = ESP.getFreeHeap();
      msg["foo"] = foo_4000;
      msg.prettyPrintTo(mqtt_buff, MQTT_MAX_PACKET_SIZE);

1TekPro, 2019-01-29:
Does it publish if you comment out foo?

macedolfm, 2019-01-29:
> @macedolfm Does it publish if you comment out foo?

Sure...this is what the MQTT server gets:

>{
  "millis" : 18541063,
  "free_mem" : 204496,
  "foo" : "01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789"
}

macedolfm, 2019-01-29:
Updates on this issue. I managed to create a multitask version of the MQTT client posting some 580 bytes, over WiFiClientSecure every 5 seconds (code bellow).

A VERY CLEAR pattern emerged from this data: EVERY 15 minutes, almost exactly, the Client x Server was broken and had to be re-connected, and I mean it, EVERY 15 minutos give or take a couple seconds, so it is very clear that some one behind the scenes was doig something in the wrong way.

The reconnection time every 15 minutes varies from an unoticeable 1 seconds delay up to 36 seconds, each attempt 4 seconds appart. Try the code and see if you get the same results:

```
const char* test_root_ca = "-----BEGIN CERTIFICATE-----\n" \
  "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
  "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
  "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
  "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
  "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
  "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
  "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
  "SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
  "Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
  "a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
  "/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
  "AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
  "CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
  "bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
  "c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
  "VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
  "ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
  "MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
  "Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
  "AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
  "uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
  "wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
  "X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
  "PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \
  "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
  "-----END CERTIFICATE-----";

const char* foo_3000 = "0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012 [... cut ...]
..O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O...
..O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O...
..O.....O.....O.....O.....O.....O.....O.....O.....O......O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O..
...O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O..
...O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....O.....

```

1TekPro, 2019-01-30:
If the task handler is NULL and all are core 1, why pinnedtocore?

macedolfm, 2019-01-30:
This was one of many tested configurations, core 1, core 2, different task priorities. Any suggestions?

⁣Sent from TypeApp ​

1TekPro, 2019-01-30:
@macedolfm Could this be related to a server timeout or "antihack/flood" on the server side?

macedolfm, 2019-01-30:
Well, it might, I will check that.

From: 1TekPro
Sent: Wednesday, January 30, 2019 9:21 AM
To: espressif/arduino-esp32
Cc: Luis F. Macedo; Mention
Subject: Re: [espressif/arduino-esp32] WifiClient & WiFiClientSecurerelaibility & disconnections when being used by PubSubClient (#2394)

@macedolfm Could this be related to a server timeout or "antihack/flood" on the server side?

macedolfm, 2019-01-30:
@1TekPro Found out some intersting answers reading over the Mosquitto log files. The ESP32 actualy DID NOT disconnected so often as it reported to the running skecth. Running another MQTTfx client on the side I could se as the ESP32 was reporting its status as mqtt.disconnected() but the MQTT server was still receiving live data from the module from a few seconds behind. When the module catched up with all the queued messages inside it the reported MQTT as connected again. So I guess this gotta be some ESP32 internal routine that exists to prevent data flooding on the module TX buffer, The server can keep up with much more than the ESP32 can provide. From my experiments I think the ESP32 can send no more then 512 bytes/second using MQTT+SSL.

1TekPro, 2019-01-30:
Just for claification, when you did the core 0 test, did you run multi core with only mqtt in core 0?

macedolfm, 2019-01-31:
Running anything related to WiFi & MQTT on the core 0 was a big NO, a sorts of erros happened. Still strugling with 15 minutes disconnection mystery, stil pretty consistent and on the clock.

From: 1TekPro
Sent: Wednesday, January 30, 2019 8:41 PM
To: espressif/arduino-esp32
Cc: Luis F. Macedo; Mention
Subject: Re: [espressif/arduino-esp32] WifiClient & WiFiClientSecurerelaibility & disconnections when being used by PubSubClient (#2394)

Just for claification, when you did the core 0 test, did you run multi core with only mqtt in core 0?

1TekPro, 2019-01-31:
And have you tried without pinning, just let it float?

macedolfm, 2019-01-31:
All sorts of crazy combinatiosn you can imagine in 3 days of battleing this issue, so far, not a clue of the way of the darkness

1TekPro, 2019-01-31:
Sherlock Holmes time... Once you eliminate the impossible, whatever remains, no matter how improbable, must be the truth.

Eliminate everything that is "impossible" to be causing the error. Example: It connects, so not your wifi code. It sends data, not the Matt. Etc...

closed 2019-08-15 by stale[bot]