espressif/arduino-esp32 #2395: BLE advertising service data

rado99, 2019-01-29:
Hello, now I am using BLE with 3 services and some charasterisitcs in it. I need to advertise some service data before connection, but when I enable it, the BLE not works, either advertising like a beacon nor to connect with it.
Here is my function for updating the service data:
```
void updateAdvertisingData(){
   BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();
  
  //oAdvertisementData.setFlags(0x04); // BR_EDR_NOT_SUPPORTED 0x04
  
  std::string strServiceData = "";
  char buff[20];
  sprintf(buff,"%.2f %.2f %c",sens.temperature,sens.humidity,sens.condition);
  strServiceData += (char)strlen(buff);     // Len
  strServiceData += (char)0xFF;   // Type
  strServiceData += buff; 
  oAdvertisementData.addData(strServiceData);
  
  beaconAdvertising->setAdvertisementData(oAdvertisementData);
  beaconAdvertising->setScanResponseData(oScanResponseData);
}
```

And here is the enabling of the services:
```
  sensor->start(); //Service
  pService->start(); //Service
  alarms->start(); //Service
BLEAdvertising *pAdvertising = pServer->getAdvertising();
   beaconAdvertising = BLEDevice::getAdvertising();
  updateAdvertisingData();
  beaconAdvertising->start();
  pAdvertising->start();
```

chegewara, 2019-01-29:
```
beaconAdvertising->start();
  pAdvertising->start();
```
There is no point to call both because esp32 can make only one advertising at the same time. Maybe it is causing some conflict, but as far as i remember second advertising start just should be ignored.

rado99, 2019-01-29:
I commented out the  `pAdvertising->start()` , but without change of the state. Here are the few debug info from the end of the bluetooth init of the device, I think I would add manual the charasteristics in beaconAdvertising, but why ever it doesn't work like a beacon now?
```
[D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:152] setAdvertisementData(): >> setAdvertisementData
[D][BLECharacteristic.cpp:203] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
[D][BLEAdvertising.cpp:160] setAdvertisementData(): << setAdvertisementData
[D][BLECharacteristic.cpp:459] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:169] setScanResponseData(): >> setScanResponseData
[D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
[E][BLEAdvertising.cpp:174] setScanResponseData(): esp_ble_gap_config_scan_rsp_data_raw: 258 Unknown ESP_ERR error
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:177] setScanResponseData(): << setScanResponseData
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 4]
[D][BLEAdvertising.cpp:186] start(): >> start: customAdvData: 1, customScanResponseData: 1
[D][BLEAdvertising.cpp:205] start(): - no services advertised
[D][BLEAdvertising.cpp:246] start(): << start
[D][BLEDevice.cpp:563] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:479] handleGAPEvent(): handleGAPEvent [event no: 6]
```

closed 2019-08-15 by stale[bot]

martimalek, 2020-05-05:
Hello, I'm having an issue when trying to advertise service data. My goal is to advertise a complete list of 128-bit service uuid and some service data at the same time. But currently i can't get them to coexist.

```
BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
BLEAdvertisementData advertisementData = BLEAdvertisementData();
BLEUUID bleUUID = BLEUUID::fromString(SERVICE_UUID);

advertisementData.setCompleteServices(bleUUID); // This sets the list of uuids
advertisementData.setServiceData(bleUUID, "Some data"); // This sets the service data
// WARNING!!! If the above two lines are used at the same time then only one of the
// two takes effect

pAdvertising->setAdvertisementData(advertisementData);
pAdvertising->setScanResponse(true);
pAdvertising->start();
```
I don't understand why this behaviour is happening. I can for example set the service data and also some flag with `advertisementData.setFlags` but i can't do it with the service data and the list of uuids. 

Any help is much appreciated.