espressif/arduino-esp32 #2396: ESP32 does not answer ARP requests

edalongeville, 2019-01-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
The ESP32 dos not answer ARP requests.

At startup, the ESP32 sends a gratuitous ARP.
Once ESP32 startup is complete, I flush the arp cache on my Ubuntu machine (sudo ip -s -s neigh flush all) and attempt to ping the ESP32. The ARP requests sent by my machine are left unanswered.

I suspect this is a known issue, as the ESP32 sends gratuitous ARP about once per minute. This works as a temporary workaround, but in some applications, ARP requests need to be answered.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <aREST.h>

// WiFi parameters
String ssid = "MYSSID";
String password= "MYPASSWORD";

#pragma region aRESTandWiFi

// The port to listen for incoming TCP connections
#define LISTEN_PORT           80
// Create an instance of the server
WiFiServer server(LISTEN_PORT);

void setup(void)
{

  // Start Serial
  Serial.begin(115200);

  delay(5000);

  // Connect to WiFi
  Serial.print("Connecting to wifi with ssid/password: " + ssid + "/" + password);
  WiFi.begin(ssid.c_str(), password.c_str());
  while (WiFi.status() != WL_CONNECTED) {
     delay(500);
     //write_to_logs("...");
  }
  Serial.print("WiFi connected");
  Serial.print("\n");
  Serial.print(WiFi.localIP()); // Print the IP address
  Serial.print("\n");

  }

void loop() {
  }
```

### Debug Messages:
```
Connecting to wifi with ssid/password: MYSSID/MYPASSWORD[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.1.39, MASK: 255.255.255.0, GW: 192.168.1.1
WiFi connected
192.168.1.39 
```

closed 2019-08-15 by stale[bot]

szundi, 2020-04-13:
Hello, this stale bot is silly. This is a problem for me to in 2020. Stale closed tickets don't fix themselves.

lbernstone, 2020-04-13:
By default, the esp32 runs in power save mode, where it lets the router queue up requests and only sends periodic beacon signals.  To turn it off, you can do `WiFi.setSleep(false);`.  I don't know if that will fix this, but it might.

nopnop2002, 2020-04-23:
Hello.

ARP inquiry by broadcast (who is 192.168.10.2?) Consumes CPU.

This is because the ARP inquiry Dessination Ethernet address is ff: ff: ff: ff: ff: ff and is broadcast.

So, All the machines in the same segment that receive this will think "maybe it's me" just by looking at the Ethernet address.

Then, look at the contents of the ARP packet, and finally discard it because you do not matter

In my opinion, ESP32 may be ignoring broadcast ARP inquiry.

As an alternative, ESP32 issues ARP announcements every 60 seconds.

nopnop2002, 2020-04-24:
This issues is very similar to the one below.

https://github.com/esp8266/Arduino/issues/6886

It can occur on certain routers such as Mikrotik and Fritzbox.

vito45h, 2020-05-06:
I have run in to a similar issue, for me my ESP32 is trying to send a UDP packet to a specific IP address.

I have a wireshark on my computer where i can see the ESP32 broadcasting an arp request and instantly the computer sends back a direct arp, as seen in the following screenshot
[image]

At this point the ESP32 can't be ping from the computer, it seems the only way to get it working again is to power cycle it.

jnwatts, 2020-06-20:
I've been struggling with this for weeks: My ESP is acting as a server, and my clients would sometimes work just fine, and other times take up to a minute (usually timing out) to get a response from the ESP. The issue is not a _failure_ to respond to ARP, it's that the wifi-modem is in _low power_ by default!!

The only reason my clients ever managed to work, was due to the default gratuitous arp frequency of 60 seconds: Client would issue ARP request, and IF a G-ARP happens within the connect-timeout period, then the connection would succeed and all is well.

Unfortunately, I'm not sure how/where to set this for arduino-esp32, as I switched over to esp-idf. In the idf API, the solution is to turn of wifi-modem sleep:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html
```
ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));
```

lbernstone, 2020-06-20:
WiFi.setSleep(false);