espressif/arduino-esp32 #2398: send_ssl_data fails on large files - UNKNOWN ERROR CODE (0050)

thezenox, 2019-01-29:
Im running multiple tasks on a custom ESP32 WROOM based datalogger. It creates files on a microSD card and uploads them via HTTP POST if WiFi is avaliable (can be multiple files in one POST). With files below 400kB everything works fine most of the time. If I try to uploading files larger than 2MB if fails at some random point (between 1.2MB and 1.8MB send).
My Blocksize is 2500 byte in this example, but blocksizes from 1300 to 8000 doesn't change anything. I also tried put a 10ms delay every 200kB.
The recorder task is priority 1 generating a sd_write buffer, all sd write oberations are handled by a write task with priority 3, the wifi_upoad task is priority 1. all sd operations are managed by a binary semaphore.

#define UL_BUFFERSIZE 2500
uint32_t size_send= 0;
uint32_t size_written =0;
uint32_t size_to_read =0;
File file;
uint8_t* buffer = new uint8_t[UL_BUFFERSIZE];
... // send HTTP Post headers and body
 while(size_written != size[i]){ // size[i] is filesize
            if(size[i]-size_written > UL_BUFFERSIZE){size_to_read=UL_BUFFERSIZE;}
            else {size_to_read= size[i]-size_written;}
            if( xSemaphoreTake( xMutex_sd, ( TickType_t ) SD_WAIT_TICKS ) == pdTRUE ){
              file = sd.open(&filepath[i][0]);
              if(!file){
                // some error handling and free() buffers
                return false;}
              file.seek(size_written);
              if(file.available() >= size_to_read){
                    file.read(buffer, size_to_read);
                    file.close();

                    xSemaphoreGive( xMutex_sd ); // release sd card for sd write process
                    size_written+= size_to_read;
                    size_send= client_secure->write((const uint8_t *)buffer, size_to_read);

                      if(size_send != size_to_read){ 
                          // error handling. this is triggered on write fail.
                        }

... // sending first blocks of data...
[V][ssl_client.cpp:274] send_ssl_data(): Writing HTTP request...
1220000 <-- This is bytes send
[V][ssl_client.cpp:274] send_ssl_data(): Writing HTTP request...
1222500
[V][ssl_client.cpp:274] send_ssl_data(): Writing HTTP request...
1225000
[V][ssl_client.cpp:274] send_ssl_data(): Writing HTTP request...
1227500
[V][ssl_client.cpp:274] send_ssl_data(): Writing HTTP request...
[E][ssl_client.cpp:33] handle_error(): UNKNOWN ERROR CODE (0050)
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -80
[V][ssl_client.cpp:243] stop_ssl_socket(): Cleaning SSL connection.

Any suggestions on what is going wrong here?

thezenox, 2019-02-13:
Seems to be a server timeout, nothing related to esp32

closed 2019-02-13 by thezenox

mirecta, 2021-03-18:
no its on esp32 side  its fixed now u can see https://github.com/espressif/arduino-esp32/issues/4960

lbernstone, 2022-11-03:
Please don't hijack closed issues.  Your problem doesn't seem related to this.  If you want help, open a new issue and follow the reporting guidelines.

hitecSmartHome, 2024-07-05:
same problem