espressif/arduino-esp32 #2399: ESP32 crashed when reconnect to another ESP32 as AP

yo90bosses, 2019-01-29:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1-rc2
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Im using two ESP32 boards. One as a Station and the other as an access point and using the asyncUDP library to braodcast UDP messages ten times a second. If i remove power from the AP then the station tries to reconnect as normal and a BEACON_TIMEOUT message comes when in Verbose debug mode. The problem comes when i press the restart button on the AP. That causes the station to disconnect but differently than before and a NOT_AUTHED message comes instead of the othet from Verbose and only sometimes when the station then reconnects it crashes. This never seems to happen when i diconnect power from the AP and the station disconnects with the BEACON_TIMEOUT istead of NOT_AUTHED.

I cant have this happen as im using the station as a flight controller and if the ESP32 crashes then i loose control of the plane.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
when i press the reset button:
```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 6 - NOT_AUTHED
```

when i remove power:
```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
```

when it crashes:
```
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
[... 64 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40149f7c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalGuru Meditation Error: Core  1 panic'ed (Double exception)
Guru Meditation Error: Core  1 panic'ed (P⸮⸮?⸮⸮⸮?)
Debug exception reason: BREAK instr 

```

atanisoft, 2019-01-29:
you haven't provided code reference, but the above usually indicates you are doing something with an interrupt that is not working as it should on the esp32

yo90bosses, 2019-01-30:
@atanisoft I added the code. It's exactly the same as in my other sketch but I removed all the things that don't have anything to do with the wifi. If your saying it has something to do with interrupts then I think it might be the asyncUDP library. The only other thing using interrupts is a ultrasound ping sensor but that has been working for weeks without a problem. Only now that I have added wifi.

1TekPro, 2019-01-30:
@atanisoft
And if wifi is not connected?

yo90bosses, 2019-01-30:
@1TekPro 
It only happens when I try to connect to wifi. 

@atanisoft
I found out that if I remove the interrupt for the ultrasonic sensor it doesn't happen anymore. 

All the variables in the interrupt routine are declared as volatile so I don't understand why the interrupt for the ultrasonic sensor is interfering with the wifi. I also changed the UDP code from the asyncUDP library to the other one.

atanisoft, 2019-01-30:
@yo90bosses then the ISR in that ultrasonic sensor code is the source of your issues. It is not written correctly. It is attempting to access flash memory from inside the ISR, ref: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spi_flash.html#concurrency-constraints

1TekPro, 2019-01-30:
Oh snap! lol

yo90bosses, 2019-01-31:
@atanisoft thanks for the help. It hasn't crashed since I moved the ultrasonic interrupt into the IRAM.

closed 2019-01-31 by yo90bosses

atanisoft, 2019-01-31:
@yo90bosses you're welcome! Glad you were able to get it resolved. I'm still working through some issues with my hw timer ISR (its in IRAM) but spi_flash is still causing some issues. I'm working on rewriting my ISR and moving even the data being used into IRAM (with some challenges)

burkulesomesh43, 2019-05-24:
I am also getting same issue. but I am using Serial communication with other MCU.
Can you please help me out for the same.