espressif/arduino-esp32 #2401: BLE refuses to connect when WiFi is connected

wezleysherman, 2019-01-29:
Right now I have my ESP32 configured to automatically connect to WiFi when it is turned on.
I'm also allowing it to be connected via BLE for additional configuration.

For some reason, when it is connected to WiFi it still advertises BLE, but when I'm trying to connect the connection will outright fail. The only want to properly connect via BLE is to reset the device and connect before it connects to WiFi.

Is there a way to detect when the ESP32 is making that initial handshake with my other devices? This way I can disconnect WiFi whenever someone attempts to connect to the ESP.

chegewara, 2019-01-30:
Hi,
i dont see any problem with my test app, but it may be because i am using a bit modified BLE library that i will push soon to 

here is my log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
Starting BLE work!
....
WiFi connected
IP address:
172.16.1.161
Starting BLE work!
Starting BLE work!
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 0
Starting BLE work!
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 7
Starting BLE work!
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 65535
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 9
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 9
[E][BLEDevice.cpp:555] getAdvertising(): create advertising
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 12
[W][sketch_jan28a.ino:33] my_gap_event_handler(): custom gap event handler, event: 0
[W][sketch_jan28a.ino:33] my_gap_event_handler(): custom gap event handler, event: 1
[W][sketch_jan28a.ino:33] my_gap_event_handler(): custom gap event handler, event: 6
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 14
[W][sketch_jan28a.ino:33] my_gap_event_handler(): custom gap event handler, event: 20
[W][sketch_jan28a.ino:33] my_gap_event_handler(): custom gap event handler, event: 20
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
Characteristic1 Read
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 1
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 21
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 1
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 21
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
Characteristic2 Read
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 1
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 21
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 42
[E][BLECharacteristicMap.cpp:87] handleGATTServerEvent(): 44
[W][sketch_jan28a.ino:41] my_gatts_event_handler(): custom gatts event handler, event: 1
[... 84 lines cut ...]
  pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC2_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new Characteristic2_Callbacks());
  pCharacteristic->setValue("I am Characteristic #2");

  Serial.println("Starting BLE work!");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

wezleysherman, 2019-01-30:
Thanks! @chegewara 

Does your sketch connect fine via BLE even when WiFi has already established a connection?

I think maybe your bits to help with iPhone connection issues may help my connection issues.

chegewara, 2019-01-30:
As you can see from logs i am not even starting BLE before i connect to wifi, but i think that makes no difference. Just wanted to test if i can connect to esp32 with ble client when esp32 is already connected to wifi:
```
WiFi connected
IP address:
172.16.1.161
Starting BLE work!
```

wezleysherman, 2019-01-30:
Yeah, since a WiFi connection on my device isn't always guaranteed I can't wait for WiFi to connect before starting BLE. 

My sketch does attempt to connect to WiFi before starting the BLE service, but as soon as WiFi establishes connection I can no longer connect to BLE (It still advertises though).

Maybe this is because I'm starting the BLE service prematurely... 

I wonder if I should scan for the WiFi network first, have a connection timeout if it sees the SSID but can't connect, and then start my BLE service. I guess my only concern at that point is the ESP32 taking too long to connect to the WiFi network and causing a short period of time where BLE is unavailable.

chegewara, 2019-01-30:
I changed order and now first i initialize BLE and all wifi connection calls are at the end of setup and still no issue with connection.

Maybe try with https://github.com/nkolban/esp32-snippets master. I just pushed code with some bugfixes.

closed 2019-02-05 by wezleysherman

vcmorini, 2019-05-15:
guys, one question.

should I merge arduino-esp32 with nkolban/esp32-snippets to fix?

thanks in advance :)

mrdc, 2019-10-22:
@chegewara 
> and my sketch:

Unfortunately, when I add simple code for Firebase communication, EPS32 reboots because of panic:
```
12:39:59.642 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
12:39:59.642 -> Core 0 register dump:
12:39:59.642 -> PC      : 0x4015d1d4  PS      : 0x00060330  A0      : 0x801572a1  A1      : 0x3ffe28f0  
12:39:59.676 -> A2      : 0x3ffe2938  A3      : 0x3fffffc0  A4      : 0x00000000  A5      : 0x00000004  
12:39:59.676 -> A6      : 0x3ffdf230  A7      : 0x3ffb79c8  A8      : 0x8015d1d2  A9      : 0x3ffe28e0  
12:39:59.676 -> A10     : 0x3ffe28f8  A11     : 0x3fffffc7  A12     : 0x00000006  A13     : 0x3ffe28fe  
12:39:59.676 -> A14     : 0x000000c1  A15     : 0x000000e9  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
12:39:59.710 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
12:39:59.710 -> 
12:39:59.710 -> Backtrace: 0x4015d1d4:0x3ffe28f0 0x4015729e:0x3ffe2930 0x4008e6a9:0x3ffe2960

```

chegewara, 2019-10-23:
This can be anything, but most likely heap/ram shortage.
Like you said, when you add Firebase communication, which has nothing to do with BLE + WiFi.

mrdc, 2019-10-23:
@chegewara 

I was able to fix it with `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);`
No more panic in many hours.

mrdc, 2019-11-09:
@chegewara 

Do you experience any delays with your code https://github.com/espressif/arduino-esp32/issues/2401#issuecomment-458777266 when using BLE?
I have noticeable delays when sending data. With this code BLE Read/Write operations are instant. Can't understand why speed is so different:

```c
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLECharacteristic *pCharacteristic;
bool deviceConnected = false;

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();
    }
};

void setup() {
  BLEDevice::init("Test"); // Give it a name
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID_TX,
                      BLECharacteristic::PROPERTY_NOTIFY
                    );
  pCharacteristic->addDescriptor(new BLE2902());
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_RX,
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
  pCharacteristic->setCallbacks(new MyCallbacks());
  pService->start();
  pServer->getAdvertising()->start();
}

void loop() {
  if (deviceConnected) {
    pCharacteristic->setValue("Hello!");
    pCharacteristic->notify(); // Send the value to the app!
  }
}
```

chegewara, 2019-11-10:
Since i posted that code i am now in so different esp32 world. Its hard to say now if i got delays or not.

mrdc, 2019-11-11:
@chegewara it looks like the issue happens when heavy code is executed during BLE work. Or it's timing issue inside SDK.
Maybe executing tasks on second core helps - have no clue as I'm novice in ESP32 world.