espressif/arduino-esp32 #2402: Ethernet won't work on core 0

nixmeer, 2019-01-30:
Hello,

I got a heavy control algorithm that is supposed to run on one core and use it's full capacity. Coprocessor communication (spi) and ethernet connection is supposed to use the other core. 

The follow code works perfectly fine:
```
  xTaskCreatePinnedToCore(
      ethernet, /* Function to implement the task */
      "Ethernet", /* Name of the task */
      10000,  /* Stack size in words */
      NULL,  /* Task input parameter */
      9,  /* Priority of the task */
      &ethernetTask,  /* Task handle. */
      1); /* Core where the task should run */
```
It prints `ETH connected` and a few moments later 

```
IPv4: 192.168.101.115
SubnetMask: 255.255.255.0
FULL_DUPLEX
100Mbps
```

If I change the core number to 0, it still prints `ETH connected` -> `WiFiEvent()` seems to be called. But it does not print the rest and the router does not show a device or an IP address.

Any idea why it is not possible to pinn the ethernet task to core 0? Or should I post more of my code?

Best

Jan

1TekPro, 2019-01-30:
Without seeing your code, I would guess you didn't create a task.

beegee-tokyo, 2019-01-31:
Try to change the task priority. I usually run tasks on core 0 with an priority below 5.

closed 2019-08-15 by stale[bot]