espressif/arduino-esp32 #2403: Arduino.h wrong #include<algorithm>

samueldinesh, 2019-01-30:
### Debug Messages:
Using board 'esp32doit-devkit-v1' from platform in folder: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.1
Using core 'esp32' from platform in folder: /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.1
....
/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.1/cores/esp32/Arduino.h:148:21: fatal error: algorithm: No such file or directory
compilation terminated.
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

##from errored file
from Arduino.h line 145 to 150
#ifdef __cplusplus
}

#include <algorithm>
#include <cmath>

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			 ?1.0.1? 
IDE name:						?Arduino IDE? 
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?921600?
Computer OS:                                      ?Ubuntu?

### Description:
How to solve this problem. Did below line solve the problem
#include <algorithm.h>
#include <cmath.h>

Please tell. Thank you

samueldinesh, 2019-01-30:
How to solve this problem. Did below line solve the problem
#include <algorithm.h>
#include <cmath.h>

There is no above file. I think esp32/core/1.0.1 is under development. I try to use 1.0.0

closed 2019-08-15 by stale[bot]

danewber, 2019-11-22:
I just updated to 1.0.4 and now I'm having this exact issue.  I can't compile for any esp32 device.

dvoelkel, 2020-08-25:
I have the same problem with 1.0.4 when compiling for ESP32.
I don't know how to fix this.

lbernstone, 2020-08-25:
algorithm is part of the toolchain.  Run [get.exe](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md) from a cmd window.