espressif/arduino-esp32 #2405: problem whit serial bt

Wailshire, 2019-01-30:
I have a problem whit Sending data over serial bluetooth . it works perfectly sendig over serial bluetooth but only if the esp32 is conected in a computer whit the serial port open. otherwise it appears on the bluetooth that the port is busy. Any idea?

Wailshire, 2019-01-30:
it's worse than i thought. when i close serial prot the esp32 don run any program. even a simple blink. have anyone solved this?

beegee-tokyo, 2019-01-30:
Can you post your setup() code. With what you say nobody can help.

Wailshire, 2019-01-30:
> Can you post your setup() code. With what you say nobody can help.

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

this simple sketch dont work if the serial port is closed.

sansillusion, 2019-01-30:
Sounds like a power supply issue to me

beegee-tokyo, 2019-01-30:
Agree with power problem. How do you power your ESP32?

closed 2019-08-15 by stale[bot]