espressif/arduino-esp32 #2410: ESP32 TASMOTA WIFI MODULE NOT ENABLED

eramanaraju, 2019-01-31:
00:00:00 APP: Set Baudrate to 115200

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_00000000, GroupTopic sonoffs) Version 5.12.0c-STAGED

Backtrace: 0x4008d4c0:0x3ffb1df0 0x4008d6f1:0x3ffb1e10 0x40090351:0x3ffb1e30 0x4013453e:0x3ffb1e70 0x4013481a:0x3ffb1e90 0x4012432c:0x3ffb1eb0 0x40124395:0x3ffb1ed0 0x4012c0d5:0x3ffb1ef0 0x4012c0fa:0x3ffb1f10 0x40122e3f:0x3ffb1f30 0x400d447a:0x3ffb1f50 0x400dfe68:0x3ffb1f70 0x400f55e3:0x3ffb1fb0 0x4008f751:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698

eramanaraju, 2019-01-31:
HI I HAVE LOADED ESP32 TASMOTA FIRMWARE IN ESP32 BOARD
I GOT THIS MESSAGE PLEASE TELL ME THE SOLUTION  FOR IT
00:00:00 APP: Set Baudrate to 115200

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_00000000, GroupTopic sonoffs) Version 5.12.0c-STAGED

Backtrace: 0x4008d4c0:0x3ffb1df0 0x4008d6f1:0x3ffb1e10 0x40090351:0x3ffb1e30 0x4013453e:0x3ffb1e70 0x4013481a:0x3ffb1e90 0x4012432c:0x3ffb1eb0 0x40124395:0x3ffb1ed0 0x4012c0d5:0x3ffb1ef0 0x4012c0fa:0x3ffb1f10 0x40122e3f:0x3ffb1f30 0x400d447a:0x3ffb1f50 0x400dfe68:0x3ffb1f70 0x400f55e3:0x3ffb1fb0 0x4008f751:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698

lbernstone, 2019-01-31:
Please read the [bug report template](https://github.com/espressif/arduino-esp32/blob/master/.github/ISSUE_TEMPLATE/bug_report.md). This forum is for issues with the arduino-esp32 codebase, not 3rd party libraries.

beegee-tokyo, 2019-01-31:
@eramanaraju right github.repo issues to report this is [here](https://github.com/JacekDob/Sonoff-Tasmota)

closed 2019-02-02 by copercini