espressif/arduino-esp32 #2411: Setting GPIO6 to OUTPUT causes Core 1 panic

hicotton02, 2019-01-31:
### Hardware:
Board:							ESP32-WROOM-32U
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						Arduino IDE 1.8.6
Flash Settings:
ESP32 Wrover Module, Default, DIO(I am using all 3 UARTs), 80MHz, 921600, Error
Computer OS:                                        Windows 10

### Description:
if i set "pinmode(6, OUTPUT);" i get:
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

if I comment it out, works fine. I created a UnitTest Sketch for this post and the error is slightly different.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
void setup() {
  // put your setup code here, to run once:
pinMode(6,OUTPUT);
Serial.begin(115200);
Serial1.begin(115200);
Serial2.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println("Test");
delay(500);
}
```

### Debug Messages:
```
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

```

everslick, 2019-01-31:
GPIO 06 to GPIO 11 are used for the internal flash chip. They are VERBOTEN. It is well documented.

hicotton02, 2019-01-31:
I thought it was just 9. so there is no way to use it? I am also using 7. it's not causing an issue, but I havent fully tested it yet.

hicotton02, 2019-01-31:
I found the documentation...

closed 2019-01-31 by hicotton02

Lokecrafter, 2022-04-06:
> GPIO 06 to GPIO 11 are used for the internal flash chip. They are VERBOTEN. It is well documented.

Ahhh after hours of trying to fix a problem where my esp32 was stuck in a reboot loop this solved it. You're my saviour @everslick !   <3