espressif/arduino-esp32 #2412: Minor code change in esp32-hal-i2c.c for arduino-esp32

Tread-Qualitrol, 2019-01-31:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
compile for m5stack, how to submit a request for code change?

Describe your system( Hardware, computer, O/S, core version, environment).
linux mint 19.1 , dell xps 8700, latest 

Describe what is failing.
Line 454 in esp32-hal-i2c.c returns a false in a void function. Causes all sorts of errors in eclipse, i don't like to disable warnings so here I am.

Show the shortest possible code that will duplicate the error.

Line 453 if(i2c == NULL) {  // point to peripheral control block does not exits
Line 454     return false;
Line 455 }
Show the EXACT error message(it doesn't work is not enough).

Return has value, in function returning void

All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							M5Stack Fire
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git?
IDE name:						Arduino IDE, IDF component, hybrid
Flash Frequency:					80 Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        Linux Mint 19.1

### Description:
Describe your problem here

Line 454 in esp32-hal-i2c.c returns a false in a void function. Causes all sorts of errors in eclipse, i don't like to disable warnings so here I am.

stickbreaker, 2019-01-31:
@Tread-Qualitrol Yes, I see the error. Why don't you generate the pull request to fix it.

It is simple:
* under the CODE tab navigate to ../cores/esp32/esp32-hal-i2c.2
* in the upper right of the file's title bar, you will see a pencil icon (edit mode, select it).
It will popup a message about editing this file in your fork.
* make the change( fixing the error)
* at the bottom of the web page you will see two fields, a Title, and a Summary.  fill them in.
* there is a green button "propose file change", select it.
* then it will popup a banner about creating a pull request, accept it

Or,
 I can do it?
Chuck.

stickbreaker, 2019-02-01:
@nae9 I think I understand the problem you describe in #2345. But, I don't use WiFi in my esp32 projects, so I don't have much experience with the WiFi components.

I recommend you create a fork of arduino-esp32, modify the examples to fix the problem, then create a pull request.  This allows others to see exactly what problems you have found.  Your solution may not be accepted but, it will start a conversation about the problem. 

I started using the esp32 in 2017, i found the i2c subsystem to be unreliable, i re-wrote it November 2017. Submitted a pull request December 2017. A highly modified re-write was merged into the code base for the July 2018 release. It take time to review and validate changes.  

The sequencing fix you propose is just covering an error deeper in the code. That deeper error is what should be identified and fixed.

Chuck

stickbreaker, 2019-02-19:
@Tread-Qualitrol fixed with [commit](https://github.com/espressif/arduino-esp32/commit/8ec76405b92ada0a5b4c365dbcd614161ff518be)

You can close this issue.

Chuck.

closed 2019-08-15 by stale[bot]