espressif/arduino-esp32 #2414: ETH(LAN8720) UDP + 32.768 Hz Crystal = Core  0 panic'ed (Interrupt wdt timeout on CPU0)

SkyTheLie, 2019-02-01:
Hardware:

Board: "Selfmade" ESP32-PICO-D4
connected with SMSC LAN8720A
and 32.768 Hz Crystal

IDE name: Arduino IDE
Flash Frequency: 80MHz
PSRAM enabled: Disabled
Upload Speed: 115200
Computer OS: Windows 10

Description:
On my board I am not using WiFi or BT. I use Ethernet to communicate with a Server via. UDP and I need to set a timestamp. That's why I use the Crystal.
The problem is if I set the RTC to 32.768 Hz Crystal and create an object from WiFiUDP udp I happen to get an exception sometime (right away or in 2-10 min).
The exception say's that: Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0).
(see below for more information).

If i use only the RTC without the communication it works perfect and otherwise to (communication without RTC). Only if i use both the exception emerges.

I set the Crystal with the software.
```
rtc_clk_32k_bootstrap(512);
rtc_clk_32k_enable(true);
rtc_clk_slow_freq_set(RTC_SLOW_FREQ_32K_XTAL);
```

For the ETH UDP communication i use the "WiFiUDPClient" and modified it with "ETH_LAN8720" example.

I attach a small version of my program where i get the same exception. Change the .txt to .ino
[small_version.txt](https://github.com/espressif/arduino-esp32/files/2820979/small_version.txt)

my program is bigger and i evaluate some data from the server. In the end it's exactly the same as the small_version

Exception decoder:
exception
>Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)
>Core 0 register dump:
>PC : 0x400835a5 PS : 0x00060034 A0 : 0x80083518 A1 : 0x3ffbe250
>A2 : 0x00000003 A3 : 0x00060023 A4 : 0x00060021 A5 : 0x00000000
>A6 : 0x00000000 A7 : 0x3ffb3e60 A8 : 0x3ff60000 A9 : 0x000004b0
>A10 : 0x3ffc2f1c A11 : 0x400866d4 A12 : 0x00000000 A13 : 0x00000001
>A14 : 0x0000000a A15 : 0x3ffbc0b0 SAR : 0x0000001e EXCCAUSE: 0x00000005
>EXCVADDR: 0x00000000 LBEG : 0x00000000 LEND : 0x00000000 LCOUNT : 0x00000000
>Core 0 was running in ISR context:
>EPC1 : 0x40142416 EPC2 : 0x00000000 EPC3 : 0x00000000 EPC4 : 0x400835a5
>
>Backtrace: 0x400835a5:0x3ffbe250 0x40083515:0x3ffbe270 0x40090b66:0x3ffbe290 0x4008ebf7:0x3ffbe2b0 0x40091919:0x3ffbe2d0 0x400817e2:0x3ffbe2e0 0x40142413:0x00000000
>
>Core 1 register dump:
>PC : 0x4008efcd PS : 0x00060034 A0 : 0x8008fe1d A1 : 0x3ffb1e60
>A2 : 0x3ffbdcc4 A3 : 0xb33fffff A4 : 0x00060023 A5 : 0x003fffff
>A6 : 0x3ffbe8e8 A7 : 0x00000001 A8 : 0x0000cdcd A9 : 0x0000abab
>A10 : 0x0000cdcd A11 : 0x00000000 A12 : 0x800909f7 A13 : 0x3ffbe870
>A14 : 0x00000008 A15 : 0x00000001 SAR : 0x00000018 EXCCAUSE: 0x00000005
>EXCVADDR: 0x00000000 LBEG : 0x4000c2e0 LEND : 0x4000c2f6 LCOUNT : 0xffffffff
>
>Backtrace: 0x4008efcd:0x3ffb1e60 0x4008fe1a:0x3ffb1e90 0x40091944:0x3ffb1eb0 0x400918fa:0x0002731d

Decoder:

>C: 0x400835a5: tick_hook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/int_wdt.c line 54
>EXCVADDR: 0x00000000
>
>Decoding stack results
>0x400835a5: tick_hook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/int_wdt.c line 54
>0x40083515: esp_vApplicationTickHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/freertos_hooks.c line 41
>0x40090b66: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2615
>0x4008ebf7: xPortSysTickHandler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 289
>0x40142413: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/pm_esp32.c line 487

Thank you for reading and spend time to help me. I hope everybody understands my problem and I appreciate for every help I could get.

lbernstone, 2019-02-01:
Have you tried moving the rtc settings to the very top of setup (and maybe put a short delay after)?  It seems likely that the ETH peripheral is using some RTC timings, and changing the timer while it is running could confuse it.

SkyTheLie, 2019-02-04:
> Have you tried moving the rtc settings to the very top of setup (and maybe put a short delay after)? It seems likely that the ETH peripheral is using some RTC timings, and changing the timer while it is running could confuse it.

Thanks for the response. 

Like this?

```
void setup() {
  rtc_clk_32k_bootstrap(512);
  rtc_clk_32k_enable(true);
  rtc_clk_slow_freq_set(RTC_SLOW_FREQ_32K_XTAL);

  delay(5000);

  // Initilize hardware serial:
  Serial.begin(115200);

  WiFi.onEvent(WiFiEvent);
  ETH.begin( PHY1 , ETH_PHY_POWER, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_PHY_LAN8720, ETH_CLK_MODE);
}
```
I tested it and it didn't help :(

closed 2019-08-15 by stale[bot]