espressif/arduino-esp32 #2417: undefined reference to esp_apptrace_vprintf() & esp_apptrace_flush()

lszlfrng, 2019-02-02:
Is the app_trace ([application trace to host](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/app_trace.html)) from esp-idf is supported in the arduino-esp32?

### Description:

I'm getting compilation error when I like to use the `esp_log_set_vprintf` or `esp_apptrace_flush` functions.

Code: [app-trace.code.txt](https://github.com/espressif/arduino-esp32/files/2824682/app-trace.code.txt)
Log: [app-trace.log.txt](https://github.com/espressif/arduino-esp32/files/2824683/app-trace.log.txt)

he code and the full build log has been attached.

### Compilation error 
...
Linking everything together...
....
... app-trace.ino.cpp.o:(.literal._Z4loopv+0x0): **undefined reference to esp_apptrace_vprintf**(char const*, __va_list_tag)
....
... app-trace.ino.cpp.o:(.literal._Z4loopv+0x10): **undefined reference to esp_apptrace_flush**(esp_apptrace_dest_t, unsigned int)
...
... app-trace.ino:12: **undefined reference to esp_apptrace_flush(esp_apptrace_dest_t, unsigned int)**
collect2.exe: error: ld returned 1 exit status

### Sketch
```cpp
#include "esp_log.h"
#include "esp_app_trace.h"

void setup() {
}

void loop() { 
  esp_log_set_vprintf(esp_apptrace_vprintf);
  ESP_LOGI(TAG, "LOG LOOP\n");
  esp_apptrace_flush(ESP_APPTRACE_DEST_TRAX, 100000);
  delay(5000);
}
```

### Hardware:

Board: ESP32-DevKitC
Core Installation version: 1.0.1
IDE name: Arduino IDE, Platform.io
Computer OS: Windows 10, Ubuntu 18.04

atanisoft, 2019-02-02:
it is not currently enabled, to use it you would need to recompile ESP-IDF

lszlfrng, 2019-02-04:
Thank you for spending time helping me!

_Can you describe in few steps to me how to recompile the esp-idf sdk?  Or does any link about that?_

Another try:
- This function works well in esp-idf. But it 
- doesn't works when I add the arduino-esp32 to the esp-idf as a component. 

It may have intentionally omitted from the arduino-esp32 building process ... ?

atanisoft, 2019-02-04:
I don't know the steps myself other than you will need to enable the feature via menuconfig.

lszlfrng, 2019-02-04:
Thank you your answer.

The feature has been enable from menuconfig and works in the esp-idf.
But with the same settings do not works when I include the arduino-esp32 as a component.

I think that your suggestion is a good way "you would need to recompile ESP-IDF". The question is: how to recompile the esp-idf in/for the arduino-esp32?

atanisoft, 2019-02-04:
@me-no-dev had written up something for it previously but I don't know the exact steps. I believe you use a generic idf project and compile with the options you want and then replace the lib file under arduino-esp32/tools/sdk.

closed 2019-08-15 by stale[bot]