espressif/arduino-esp32 #2421: Where can I securely store my Wifi connection settings?

DeltaVetal26, 2019-02-04:
Hi,

There is an OTA HTTP firmware update in my project. I want to implement a function in which the Wifi network settings will be stored in memory, which will not be reset after the OTA update. Initially, I wanted to save the settings in the EEPROM memory, but the recent [topic](https://github.com/espressif/arduino-esp32/issues/2379) of resetting the EEPROM after several weeks of work led me astray. There I heard about NVS.

The question is, where is it safe to store and update Wifi connection data? Are there any examples for this?

bertmelis, 2019-02-04:
you could put a config file in spiffs. plenty of examples for that.

DeltaVetal26, 2019-02-05:
@bertmelis , thank you for the response!

My code:
```cpp
#include "FS.h"
#include "SPIFFS.h"

/* You only need to format SPIFFS the first time you run a
   test or else use the SPIFFS plugin to create a partition
   https://github.com/me-no-dev/arduino-esp32fs-plugin */
#define FORMAT_SPIFFS_IF_FAILED true

void readFile(fs::FS &fs, const char * path) {
    Serial.printf("Reading file: %s\r\n", path);

    File file = fs.open(path);
    if(!file || file.isDirectory()) {
        Serial.println("- failed to open file for reading");
        return;
    }

    Serial.println("- read from file:");
    while(file.available()){
        Serial.write(file.read());
    }
}

void writeFile(fs::FS &fs, const char * path, const char * message) {
    Serial.printf("Writing file: %s\r\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("- failed to open file for writing");
        return;
    }
    if(file.print(message)) {
        Serial.println("- file written");
    } else {
        Serial.println("- frite failed");
    }
}

void appendFile(fs::FS &fs, const char * path, const char * message) {
    Serial.printf("Appending to file: %s\r\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("- failed to open file for appending");
        return;
    }
    if(file.print(message)) {
        Serial.println("- message appended");
    } else {
        Serial.println("- append failed");
    }
}

void setup() {
    Serial.begin(115200);
    if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)) {
        Serial.println("SPIFFS Mount Failed");
        return;
    }

    writeFile(SPIFFS, "/wifi.txt", "SSIDNAME\n"); // 
    appendFile(SPIFFS, "/wifi.txt", "PASSWORD\n");
    readFile(SPIFFS, "/wifi.txt");
    Serial.println( "Test complete" );
}

void loop() {
}
```

This code its works, but I can't figure out how to divide the answer into two variables (SSID, PASSWORD)?

bertmelis, 2019-02-05:
Like so for example: https://arduinojson.org/v5/example/config/

CelliesProjects, 2019-02-09:
In my opinion NVS is the perfect place to store login and other per-device info.

Save it like:
````
preferences.putString("WIFI_SSID", "MYPASSWORD" );
````
and retrieve it like:
````
preferences.getString( "WIFI_SSID", "DEFAULT_VALUE" );
````
[Checkout the example.](https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/examples/StartCounter/StartCounter.ino)

DeltaVetal26, 2019-02-09:
@bertmelis , @CelliesProjects 

Thank you very much! In one project I will use spiffs, and in another NVS

closed 2019-02-09 by DeltaVetal26