espressif/arduino-esp32 #2423: Problem with oled display

gigirizzi, 2019-02-04:
i try connect my esp32S with oled 0,96 but not show nothing;

#include <Wire.h>
#include "SSD1306Wire.h" 
 
SSD1306Wire  display(0x3c, 21, 22);
 
void setup() {
  Serial.begin(9600);
 Serial.println("sss");
  display.init();
 
}
 
void loop() {
  
   display.clear();
  display.drawString(0,0, "Counter: " );
  display.display();
 
  delay(5000);}

stickbreaker, 2019-02-04:
@gigirizzi try my modified repo [ssd1306](https://github.com/stickbreaker/SSD1306) a modified version a adafruits library.  It is adjusted for a 128x64 I2C oled (SCL=15,SDA=4,RESET=16) you will have to adjust it for your board.
 
```
#ifndef OLED_RST
  #define OLED_RST 16
#endif
#ifndef OLED_SDA
  #define OLED_SDA 4
#endif
#ifndef OLED_SCL
  #define OLED_SCL 15
#endif

SSD1306 display(OLED_RST);
```

Chuck.

gigirizzi, 2019-02-04:
how use OLED_RST ? i haven't rst in oled display

stickbreaker, 2019-02-04:
@gigirizzi if you have an i2c OLED it is connected with 5 wires, Ground, VCC(3.3v), SDA, SCL, RST.
IF you have a SPI version it has 7 wires, Ground, VCC(3.3v) MOSI, MISO,SCK, SS, RST.

I have a HelTec LoRa 32 with OLED, the OLED is connected to 4(SDA), 15(SCL), 16(RST).  

`SSD1306 display(OLED_RST);` Just passes the reset pin to the display Object. When my example starts the display with `display.begin()` it uses this pin to reset the OLED hardware.

```c++
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h> // all the graphic primitives 
#include <SSD1306.h>

#ifndef OLED_RST
  #define OLED_RST 16
#endif
#ifndef OLED_SDA
  #define OLED_SDA 4
#endif
#ifndef OLED_SCL
  #define OLED_SCL 15
#endif

SSD1306 display(OLED_RST);

void setup()   {                
  Serial.begin(115200); // debug to serial monitor
  Wire.begin(OLED_SDA,OLED_SCL,100000);   //initialize i2c interface using pins attached to OLED
  // by default, we'll generate the high voltage from the 3.3v line internally! (neat!)
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3D (for the 128x64)
  // init done
  uint32_t tick = millis();
  // Show image buffer on the display hardware.
  // Since the buffer is intialized with an Adafruit splashscreen
  // internally, this will display the splashscreen.
  display.display();
  delay(2000);

  // Clear the buffer.
  display.clearDisplay();

  // draw a single pixel
  display.drawPixel(10, 10, WHITE);
  // Show the display buffer on the hardware.
  // NOTE: You _must_ call display after making any drawing commands
  // to make them visible on the display hardware!
  display.display();
  delay(2000);
  display.clearDisplay();
}

void loop(){}
```

Chuck.

gigirizzi, 2019-02-04:
I have JZK 0.96 pollici interfaccia I2C SPI IIC 128x64 display OLED LCD LED modulo seriale per per Arduino, Raspberry Pi, 51 serie, MSP430 serie, STIM32 / 2, chip SCR ecc https://www.amazon.it/dp/B06XQ9J6Z6/ref=cm_sw_r_cp_api_i_RymwCbAJF68V6
With 4 pin 

Inviato da iPhone

stickbreaker, 2019-02-04:
@gigirizzi ok, then change the object creation statement to:
```c++
SSD1306 display();
```

Everything else stays the same, well, use the correct pin numbers in the `Wire.begin()` statement.

```c++
// replaced OLED_SDA with the pin number you connected SDA to.
// replace OLED_SCL with the pin number you connected SCL to.
Wire.begin(OLED_SDA,OLED_SCL,100000);   //initialize i2c interface using pins attached to OLED
  
```

Chuck.

nae9, 2019-02-20:
Make pin redefinition like 
#define SDA 4
#define SDL 15
or any other pin where you connect OLED.

closed 2019-08-15 by stale[bot]