espressif/arduino-esp32 #2424: FSBrowser /edit returns FileNotFound

GeorgeFlorian, 2019-02-05:
Hello there ! I've been tinkering with the Arduino IDE + ESP32 for the past week.

I tried the FSBrowser + AutoConnect from [here](https://github.com/Hieromon/AutoConnect/blob/master/examples/FSBrowser/FSBrowser.ino) but with no luck.

> Hardware:
> 
> Board: ESP32 DEVKIT1
> 
> Core Installation version: 1.0.1-git this one
> 
> IDE name: Arduino IDE
> 
> Flash Frequency: 80Mhz
> 
> PSRAM enabled: no
> 
> Upload Speed: 115200
> 
> Computer OS: Linux Mint 19.1 Mate
> 

This is my sketch:

```
/*
  FSWebServer - Example WebServer with SPIFFS backend for esp8266
  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the ESP8266WebServer library for Arduino environment.
*/

#include <FS.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <SPIFFS.h>
#include <AutoConnect.h>

#define DBG_OUTPUT_PORT Serial
#define LED 2

const char* ssid = "wifi-ssid";
const char* password = "wifi-password";
const char* host = "esp32fs";

WebServer server(80);

AutoConnect portal(server);
//holds the current upload
File fsUploadFile;

bool startCP(IPAddress ip) {
    digitalWrite(LED, HIGH);
    DBG_OUTPUT_PORT.println("Captive Portal started ! IP: " + WiFi.localIP().toString());
    return true;
  }

//format bytes
String formatBytes(size_t bytes){
  if (bytes < 1024){
    return String(bytes)+"B";
  } else if(bytes < (1024 * 1024)){
    return String(bytes/1024.0)+"KB";
  } else if(bytes < (1024 * 1024 * 1024)){
    return String(bytes/1024.0/1024.0)+"MB";
  } else {
    return String(bytes/1024.0/1024.0/1024.0)+"GB";
  }
}

[... 268 lines cut ...]
> 10:27:08.444 -> [AC] 192.168.1.103/_ac
> 10:27:08.477 -> [PB] at leaving build: 259664 free
> 10:27:08.477 -> [PB] Res:200, Chunked:0
> 10:27:08.477 -> [PB] Free heap:262440, content len.:6643
> 10:27:08.477 -> [PB] Transfer-Encoding:chunked
> 10:27:08.477 -> [E][WiFiClient.cpp:346] write(): 104
> 10:27:13.455 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
> 10:27:13.854 -> [AC] esp32fs.local/favicon.ico
> 10:27:13.854 -> [AC] Page[] allocated
> 10:27:13.854 -> [E][WebServer.cpp:602] _handleRequest(): request handler not found
> 10:27:13.854 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
> 10:27:18.866 -> [AC] esp32fs.local/_ac
> 10:27:18.866 -> [AC] Page[/_ac] allocated
> 10:27:18.866 -> [AC] 192.168.1.103/_ac
> 10:27:18.899 -> [PB] at leaving build: 259372 free
> 10:27:18.899 -> [PB] Res:200, Chunked:0
> 10:27:18.899 -> [PB] Free heap:262132, content len.:6643
> 10:27:18.899 -> [PB] Transfer-Encoding:chunked
> 10:27:18.899 -> [E][WiFiClient.cpp:346] write(): 104
> 

As you can see I've uploaded 4 files into the ESP32's FlashMemory. I would like to "edit" them. Maybe delete one of them, or create a new file. I can't do that since /edit does nothing.

Do you have any idea ?
Thank you !

lbernstone, 2019-02-05:
There is a [version](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/FSBrowser/FSBrowser.ino) included in the core distribution.  Give that a try.

GeorgeFlorian, 2019-02-05:
> There is a [version](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/examples/FSBrowser/FSBrowser.ino) included in the core distribution. Give that a try.

There wasn't much to modify. But it broke everything. It can't read from the Flash Memory. It's like I have nothing in memory. But I have 8 files. I've uploaded them twice. I'm using ESP32 Sketch Data Upload. And it always worked.

/edit returns FileNotFound
/list?dir= returns [ ]

```
/*
  FSWebServer - Example WebServer with FS backend for esp8266/esp32
  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the WebServer library for Arduino environment.
  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.
  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.
  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
  upload the contents of the data folder with MkSPIFFS Tool ("ESP32 Sketch Data Upload" in Tools menu in Arduino IDE)
  or you can upload the contents of a folder if you CD in that folder and run the following command:
  for file in `ls -A1`; do curl -F "file=@$PWD/$file" esp32fs.local/edit; done
  access the sample web page at http://esp32fs.local
  edit the page by going to http://esp32fs.local/edit
*/
#include <FS.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <AutoConnect.h>

#define FILESYSTEM SPIFFS
#define FORMAT_FILESYSTEM true
#define DBG_OUTPUT_PORT Serial

#define LED 2

#if FILESYSTEM == FFat
#include <FFat.h>
#endif
#if FILESYSTEM == SPIFFS
#include <SPIFFS.h>
#endif

const char* ssid = "wifi-ssid";
const char* password = "wifi-password";
const char* host = "esp32fs";

WebServer server(80);
AutoConnect portal(server);

//holds the current upload
File fsUploadFile;

bool startCP(IPAddress ip) {
    digitalWrite(LED, HIGH);
    DBG_OUTPUT_PORT.println("Captive Portal started ! IP: " + WiFi.localIP().toString());
    return true;
  }

//format bytes
String formatBytes(size_t bytes) {
  if (bytes < 1024) {
    return String(bytes) + "B";
  } else if (bytes < (1024 * 1024)) {
[... 237 lines cut ...]
  pinMode(LED,OUTPUT);
  digitalWrite(LED, LOW);
  portal.onDetect(startCP);
  
  if (portal.begin()) {
    digitalWrite(LED, LOW);
  }

  DBG_OUTPUT_PORT.println("HTTP server started");
  
  DBG_OUTPUT_PORT.println("");
  DBG_OUTPUT_PORT.print("Connected! IP address: ");
  DBG_OUTPUT_PORT.println(WiFi.localIP());

  MDNS.begin(host);
  DBG_OUTPUT_PORT.print("Open http://");
  DBG_OUTPUT_PORT.print(host);
  DBG_OUTPUT_PORT.println(".local/edit to see the file browser");

}

void loop(void) {
  portal.handleClient();
}
```

MarkyAD, 2019-02-05:
In the version you posted just above, its setting #define FORMAT_FILESYSTEM true ao its formatting your filesystem every time it runs.

GeorgeFlorian, 2019-02-05:
Why is ``` #define FORMAT_FILESYSTEM true``` there then ? Should I only use it once then comment the line ?

MarkyAD, 2019-02-05:
Set it to false, reflash sketch, re-run sketch data upload.

GeorgeFlorian, 2019-02-08:
> Set it to false, reflash sketch, re-run sketch data upload.

It still doesn't work ! I don't know why you tried closing this issue !

MarkyAD, 2019-02-08:
Dude, no one is closing any issues. @lbernstone simply put up a pull request that fixes the issue with it formatting the filesystem.

closed 2019-02-14 by me-no-dev

mattncsu, 2020-01-23:
> In the version you posted just above, its setting #define FORMAT_FILESYSTEM true so its formatting your filesystem every time it runs.

That got me too on this example

PR #2425 (2019-02-14): default FORMAT_FILESYSTEM to false in FSBrowser.ino
Fixes #2424