espressif/arduino-esp32 #2430: Serious problem with millis()

Jeroen88, 2019-02-06:
### Hardware:
Board:							ESP32 Dev Module nodeMCU 32s
Core Installation version:			latest git
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:				       default speed
Computer OS:                                        Ubuntu

### Description:
millis() is not working as expected, causing time out problems every 72 minutes in many functions of the core and user programs

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Issue #2330 was closed, but the issue is not solved in the mentioned release. In this release the type of micros() and millis() is changed to unsigned long, but this is not solving the issue!

In my honest opinion, this is a very serious error!

The problem is not fixed by declaring micros() and millis() unsigned long.

The problem is that many timeout loops depend on 2’s complement arithmetic and millis() being defined as:

unsigned long IRAM_ATTR millis()
{
    return (unsigned long) (esp_timer_get_time() / 1000);
}

See https://github.com/espressif/arduino-esp32/blob/25a2d94d636fbbe2a554a827fa6ef4d7c8692259/cores/esp32/esp32-hal-misc.c#L109

Take for instance in Stream.cpp timedRead https://github.com/espressif/arduino-esp32/blob/86fdb5b23dbc90a37ffa140dadf22b599fd9913f/cores/esp32/Stream.cpp#L31:

int Stream::timedRead()
{
    int c;
    _startMillis = millis();
    do {
        c = read();
        if(c >= 0) {
            return c;
        }
    } while(millis() - _startMillis < _timeout);
    return -1;     // -1 indicates timeout
}

The default _timeout is 1000 ms.

Now assume that esp_timer_get_time() is about to roll over at the moment that _startMillis() = millis() is assigned. So assume esp_timer_get_time() is 0xFFFFFFFF, and millis() will be 0xFFFFFFFF / 1000 is 4294967295 / 1000 is 4294967. Thus _startMillis will be 4294967.

Next assume that one do { } while loop takes 100 microseconds (the exact duration is not important to demonstrate the effect.

At the time of the evaluation of the while condition, micros() will be 0xFFFFFFFF + 100 is 99, because of 2’s complement arithmetic. So millis() will be 99 / 1000 is 0.

The while condition at the loop will evaluate to 0 - 4294967 is -4294967. Also because of 2’s complement arithmetic this evaluates to 4290672329. The while condition evaluates to 4290672329 < 1000, which is obviously false. So the while loop stops after only 100 microseconds of waiting, instead of the intended 1000 milliseconds!!!!

Stream is in the heart of many programs. And this is just one example. Many timeout loops use the same structure: millis() - start < timeout

bertmelis, 2019-02-06:
Is it a solution to implement esp8266's logic into esp32? Only difference is esp8266 starts from a uint32_t system timer whereas esp32 has a int64_t timer; both microseconds though.

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring.c#L64

stickbreaker, 2019-02-06:
@Jeroen88 try changing the equation of `millis()`, define 1000 as a long long. 
```c++
unsigned long IRAM_ATTR millis()
{
    return (unsigned long) (esp_timer_get_time() / 1000LL);
}
```

Chuck.

bertmelis, 2019-02-06:
isn't that conversion implicit?

stickbreaker, 2019-02-07:
The compiler can choose long, because output type is long, one param is long, and processer natural is long. Converting to long first is more efficient. One conversion, natural division, no output conversion.

Don't allow it the choice.

Chuck.

Jeroen88, 2019-02-07:
@bertmelis, @stickbreaker Bert, Chuck thnx for the reactions!! 

I read the extensive comment at the ESP8266 version of millis() [here ](https://github.com/esp8266/Arduino/blob/f706c83b66261df58b7d6148e343fd21cbd77206/cores/esp8266/core_esp8266_wiring.cpp.) This uses some "magic" algorithm to do a fast division. It also states that it uses the full 32 bit range for the millis() counter (see [here](https://github.com/esp8266/Arduino/blob/f706c83b66261df58b7d6148e343fd21cbd77206/cores/esp8266/core_esp8266_wiring.cpp#L77)).

The problem is that the ESP32 does not use the full 32 bit range, that is causing the issue. I tested the example code with an ESP8266 and there it works correctly.

@stickbreaker I don't think explicitly changing the type to LL will help. The problem is that the source counter _micros()_ uses the full 32 bits causing the 'trick' to substract the _start_Micros_ from micros() to always work because of the 2's complement arithmetic. However, because of the devision, _millis()_ is not using the full 32 bits, causing trouble around the moment of overflowing of the _micros()_

bertmelis, 2019-02-07:
So the esp8266 implementation should work? (after masking 64-bit micros value with 0xFFFFFFFF to get lower 32-bit part)

Issue is indeed the mix of decimal math with binary rollover logic.

Jeroen88, 2019-02-07:
@bertmelis I think it will work, and I think both the ESP32 and the ESP8266 use a unsigned long system timer, so masking should not be necessary

bertmelis, 2019-02-07:
it is signed according to this: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/esp_timer.html#_CPPv218esp_timer_get_timev

Masking to avoid compiler warnings (storing 64 bit into 32).

stickbreaker, 2019-02-07:
@Jeroen88 
>@stickbreaker I don't think explicitly changing the type to LL will help. The source counter _micros()_ uses the full 32 bits causing the 'trick' to substract the _start_Micros_ from micros() to always work because of the 2's complement arithmetic. However, because of the division, _millis()_ is not using the full 32 bits, causing trouble around the moment of overflowing of the _micros()_

Just try modifying the `millis()` code.  

>Now assume that esp_timer_get_time() is about to roll over at the moment that _startMillis() = millis() is assigned. So assume esp_timer_get_time() is 0xFFFFFFFF, and millis() will be 0xFFFFFFFF / 1000 is 4294967295 / 1000 is 4294967. Thus _startMillis will be 4294967.

`int64_t esp_timer_get_time()` is microseconds since the timer subsystem started,(approximately boot time).  2**63ms =292271..... YEARS. `int64_t` overflow is irrelevant. `esp_timer_get_time()` will always be POSITIVE.   

The problem you are seeing is the early conversion from `int64_t` to `uint32_t` just force the compiler to do what you want.  Use the full resolution of the timer as input to the division. 

Chuck.

Jeroen88, 2019-02-07:
@stickbreaker Chuck, I believe you are right! Thanks for your insights. I have no access to my systems now, but will try it this weekend and, if succesful, make a PR.

bertmelis, 2019-02-07:
On Ideone.com:
```
#include <iostream>
#include <stdint.h>
using namespace std;

int main() {
	int64_t systemTimer = 0xFFFFFFFF;
	uint32_t lastMillis = (unsigned long) (systemTimer / 1000LL);
	systemTimer += 10; // 10 µsecs later...
	uint32_t currentMillis = (unsigned long) (systemTimer / 1000LL);
	
	if (currentMillis - lastMillis > 1000UL) {
		cout << "This shouldn't happen!\n";
	} else {
		cout << "All ok!\n";
	}
	
	systemTimer += 1000000; // 1000 msecs later...
	currentMillis = (unsigned long) (systemTimer / 1000LL);
	if (currentMillis - lastMillis > 1000UL) {
		cout << "1000 msecs barely  passed.\n";
	} else {
		cout << "corner case\n";
	}
	
	systemTimer += 1000; // 1 msecs later...
	currentMillis = (unsigned long) (systemTimer / 1000LL);
	if (currentMillis - lastMillis > 1000UL) {
		cout << "1001 msecs passed\n";
	} else {
		cout << "coups\n";
	}
	return 0;
}
```
returns:
```
All ok!
corner case
1001 msecs passed
```

Jeroen88, 2019-02-07:
@bertmelis :), thnx! I think @stickbreaker suggestion will indeed work. I will make a PR this weekend

devyte, 2019-02-07:
I haven't looked at this core's code, and I haven't read this discussion in detail, so I don't know if this helps, but I thought to mention it anyways:
- Please look at the millis() implementation of the ESP8266 core. There was a long discussion there, and the current code has been unit tested, proven to be correct without drift (the /1000 is an approximation) and optimized to avoid 64bit calculations (64bit types caused over a 4X execution time increase of millis() over the original code).
- Peripherally related, please look at PolledTimeout.h in the ESP8266 core. It encapsulates the correct way to do millis() based timing calculations, including correctness for rollover. It is also unit tested. There are examples as well to showcase usage.

Jeroen88, 2019-02-08:
@devyte 
- The ESP8266 millis() uses 'magic numbers' and stores the overflow of the 32 bit counter to overcome the problem of drift. The ESP32 timer seems to be a 64 bit natively. So I expect following @stickbreaker 's suggestion to just divide by **1000LL** (or maybe even better **1000ULL**) will be correct. As far as I understand it, gcc always uses magic numbers to do divisions, so I expect the code esp_timer_get_time() / 1000ULL _to be optimized by the compiler already using magic numbers!!_
- PolledTimeout.h uses the 'normal' timeout check: return (t - _start) >=  _timeout, see [here](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/PolledTimeout.h#L92)

Jeroen88, 2019-02-08:
Solved by PR #2438

closed 2019-02-08 by Jeroen88