espressif/arduino-esp32 #2432: xgzp 6867 i2c failed to upload sketch to wemos lolin 32

Bryanbaring, 2019-02-07:
### Hardware:
Board:							wemos lolin esp32
Core Installation version:			?1.0.0
IDE name:						?Arduino IDE 1.8.5
Flash Frequency:					?40Mhz

Upload Speed:					?115200?
Computer OS:                                        ?Windows 10

### Description:
ibeen stugling for 3mmnths and submission of output is near .my wemos lolin is working find in any file example in arduino ide.. im really sure that i installed properly the esp 32 to arduino ide.. my problem is when im uploading the sketch of my xgzp 6867 pressure sensor . "im getting error compiling sketch for wemos lolin32".. or just dont write it the right . please corrrect my sketch on how to compile it in my wemos lolin 32.. thank you in advance.

 

//Change the code below by your sketch

#include "LCD1602.h"
#include "Stdio.h"
#include "XGZPC.h"

int main(void)
{
	char DisplayBuff[20];
	float XGZPC_Data = 100;

	LCD_Init();
	if(!XGZPC_Init())
		while(1);
	

	
	LCD_Write_String(0,0,"ZPC:         KPa");
	while(1)
	{

		
		}
    if(Get_XGZPC_Value(&XGZPC_Data))
		{
		  sprintf(DisplayBuff,"%6.2f",XGZPC_Data);
		  LCD_Write_String(5,0,DisplayBuff);
		}
		else
		{
			LCD_Write_String(6,0,"error");
		}
	}
}

[XGZP6867_Pressure_Sensor_Module[1].pdf](https://github.com/espressif/arduino-esp32/files/2841547/XGZP6867_Pressure_Sensor_Module.1.pdf)

this are the source code for the main ;
#include "XGZPC.h"
#include <intrins.h>

#define  _Nop()  _nop_()        //¶¨Òå¿ÕÖ¸Áî

sbit I2C_SDA = P1^4;
sbit I2C_SCL = P1^3;

xdata  char I2C_DataBuff[6];

//static void I2C_DelayUs2x(unsigned char t)
//{   
// while(--t);
//}

//static void I2C_DelayMs(unsigned char t)
//{
//     
// while(t--)
// {
//     //´óÖÂÑÓÊ±1mS
//     I2C_DelayUs2x(245);
//     I2C_DelayUs2x(245);
// }
//}

/*------------------------------------------------
                    Æô¶¯×ÜÏß
------------------------------------------------*/
static void Start_I2c()
{
  I2C_SDA=1;   //·¢ËÍÆðÊ¼Ìõ¼þµÄÊý¾ÝÐÅºÅ
  _Nop();
  I2C_SCL=1;
  _Nop();    //ÆðÊ¼Ìõ¼þ½¨Á¢Ê±¼ä´óÓÚ4.7us,ÑÓÊ±
  _Nop();
  _Nop();
  _Nop();
  _Nop();    
  I2C_SDA=0;     //·¢ËÍÆðÊ¼ÐÅºÅ
  _Nop();    //ÆðÊ¼Ìõ¼þËø¶¨Ê±¼ä´óÓÚ4¦Ì
  _Nop();
  _Nop();
  _Nop();
  _Nop();       
  I2C_SCL=0;    //Ç¯×¡I2C×ÜÏß£¬×¼±¸·¢ËÍ»ò½ÓÊÕÊý¾Ý
  _Nop();
  _Nop();
}
/*------------------------------------------------
                    ½áÊø×ÜÏß
------------------------------------------------*/
static void Stop_I2c()
{
  I2C_SDA=0;    //·¢ËÍ½áÊøÌõ¼þµÄÊý¾ÝÐÅºÅ
  _Nop();   //·¢ËÍ½áÊøÌõ¼þµÄÊ±ÖÓÐÅºÅ
  I2C_SCL=1;    //½áÊøÌõ¼þ½¨Á¢Ê±¼ä´óÓÚ4¦Ì
  _Nop();
  _Nop();
  _Nop();
  _Nop();
  _Nop();
  I2C_SDA=1;    //·¢ËÍI2C×ÜÏß½áÊøÐÅºÅ
  _Nop();
  _Nop();
  _Nop();
  _Nop();
}

/*----------------------------------------------------------------
·¢ËÍÊý¾Ý
------------------------------------------------------------------*/
static unsigned char I2C_SendByte(unsigned char Byte)
{
 unsigned char i,I2C_ACK; 
 
 for(i=0;i<8;i++)  //Òª´«ËÍµÄÊý¾Ý³¤¶ÈÎª8Î»
    {
     if((Byte<<i)&0x80)I2C_SDA=1;   //ÅÐ¶Ï·¢ËÍÎ»
       else  I2C_SDA=0;                
     _Nop();
     I2C_SCL=1;               //ÖÃÊ±ÖÓÏßÎª¸ß£¬Í¨Öª±»¿ØÆ÷¿ªÊ¼½ÓÊÕÊý¾ÝÎ»
      _Nop(); 
      _Nop();             //±£Ö¤Ê±ÖÓ¸ßµçÆ½ÖÜÆÚ´óÓÚ4¦Ì
      _Nop();
      _Nop();
      _Nop();         
     I2C_SCL=0; 
    }
    
    _Nop();
    _Nop();
    I2C_SDA=1;               //8Î»·¢ËÍÍêºóÊÍ·ÅÊý¾ÝÏß£¬×¼±¸½ÓÊÕÓ¦´ðÎ»
    _Nop();
    _Nop();   
    I2C_SCL=1;
    _Nop();
    _Nop();
    _Nop();
    if(I2C_SDA==1)
      I2C_ACK=0;     
       else 
      I2C_ACK=1;        //ÅÐ¶ÏÊÇ·ñ½ÓÊÕµ½Ó¦´ðÐÅºÅ
    I2C_SCL=0;
    _Nop();
    _Nop();
    return I2C_ACK;
}

/*----------------------------------------------------------------
¶ÁÈ¡Êý¾Ý 
------------------------------------------------------------------*/  
static unsigned char  I2C_RcvByte()
{
  unsigned char retc;
  unsigned char i;
  
  retc=0; 
  I2C_SDA=1;             //ÖÃÊý¾ÝÏßÎªÊäÈë·½Ê½
  for(i=0;i<8;i++)
      {
        _Nop();           
        I2C_SCL=0;       //ÖÃÊ±ÖÓÏßÎªµÍ£¬×¼±¸½ÓÊÕÊý¾ÝÎ»
        _Nop();
        _Nop();      //Ê±ÖÓµÍµçÆ½ÖÜÆÚ´óÓÚ4.7us
        _Nop();
        _Nop();
        _Nop();
        I2C_SCL=1;       //ÖÃÊ±ÖÓÏßÎª¸ßÊ¹Êý¾ÝÏßÉÏÊý¾ÝÓÐÐ§
        _Nop();
        _Nop();
        retc=retc<<1;
        if(I2C_SDA==1)retc=retc+1; //¶ÁÊý¾ÝÎ»,½ÓÊÕµÄÊý¾ÝÎ»·ÅÈëretcÖÐ
        _Nop();
        _Nop(); 
      }
  I2C_SCL=0;    
  _Nop();
  _Nop();
  return(retc);
}

/*----------------------------------------------------------------
                     Ó¦´ð×Óº¯Êý
----------------------------------------------------------------*/
static void Ack_I2c(void)
{
  
  I2C_SDA=0;     
  _Nop();
  _Nop();
  _Nop();      
  I2C_SCL=1;
  _Nop();
  _Nop();              //Ê±ÖÓµÍµçÆ½ÖÜÆÚ´óÓÚ4¦Ì
  _Nop();
  _Nop();
  _Nop();  
  I2C_SCL=0;               //ÇåÊ±ÖÓÏß£¬Ç¯×¡I2C×ÜÏßÒÔ±ã¼ÌÐø½ÓÊÕ
  _Nop();
  _Nop();    
}
/*----------------------------------------------------------------
                     ·ÇÓ¦´ð×Óº¯Êý
----------------------------------------------------------------*/
static void NoAck_I2c(void)
{
  
  I2C_SDA=1;
  _Nop();
  _Nop();
  _Nop();      
  I2C_SCL=1;
  _Nop();
  _Nop();              //Ê±ÖÓµÍµçÆ½ÖÜÆÚ´óÓÚ4¦Ì
  _Nop();
  _Nop();
  _Nop();  
  I2C_SCL=0;                //ÇåÊ±ÖÓÏß£¬Ç¯×¡I2C×ÜÏßÒÔ±ã¼ÌÐø½ÓÊÕ
  _Nop();
  _Nop();    
}

/*----------------------------------------------------------------
                    ÏòÓÐ×ÓµØÖ·Æ÷¼þ·¢ËÍ¶à×Ö½ÚÊý¾Ýº¯Êý               
----------------------------------------------------------------*/
bit I2C_SendByteLen(unsigned char DeviceAddress,unsigned char RegAddress,unsigned char *s,unsigned char DataLen)
{
   unsigned char i;

   Start_I2c();               //Æô¶¯×ÜÏß
   if(I2C_SendByte(DeviceAddress) == 0)             //·¢ËÍÆ÷¼þµØÖ·
      return(0);
   if(I2C_SendByte(RegAddress) == 0)            //·¢ËÍÆ÷¼þ×ÓµØÖ·
      return(0);

   for(i=0;i<DataLen;i++)
    {   
     if(I2C_SendByte(*s) == 0)            //·¢ËÍÊý¾Ý
        return(0);
     s++;
    } 
 Stop_I2c();                  //½áÊø×ÜÏß
  return(1);
}

/*----------------------------------------------------------------
                    ÏòÓÐ×ÓµØÖ·Æ÷¼þ¶ÁÈ¡¶à×Ö½ÚÊý¾Ýº¯Êý               
----------------------------------------------------------------*/
bit I2C_RecvByteLen(unsigned char DeviceAddress,unsigned char RegAddress,unsigned char *s,unsigned char DataLen)
{
   unsigned char i;

   Start_I2c();               //Æô¶¯×ÜÏß
   if(I2C_SendByte(DeviceAddress) == 0)             //·¢ËÍÆ÷¼þµØÖ·
      return(0);
   if(I2C_SendByte(RegAddress) == 0)            //·¢ËÍÆ÷¼þ×ÓµØÖ·
      return(0);

   Start_I2c();
   if(I2C_SendByte(DeviceAddress+1) == 0)
      return(0);

   for(i=0;i<DataLen-1;i++)
    {   
     *s=I2C_RcvByte();              //·¢ËÍÊý¾Ý
      Ack_I2c();                //·¢ËÍ¾Í´ðÎ» 
     s++;
    } 
   *s=I2C_RcvByte();
    NoAck_I2c();                 //·¢ËÍ·ÇÓ¦Î»
   Stop_I2c();                    //½áÊø×ÜÏß
  return(1);
}

unsigned char XGZPC_Init(void)
{
 I2C_SDA = 1;
 I2C_SCL = 1;
 if(I2C_RecvByteLen(0xda,0xa5,I2C_DataBuff,1) == 0)
    return 0;
    I2C_DataBuff[0] &= 0xfd;
 if(I2C_SendByteLen(0xda,0xa5,I2C_DataBuff,1) == 0)
    return 0;
  I2C_DataBuff[0] = 0x0a;
  if(!I2C_SendByteLen(0xda,0x30,I2C_DataBuff,1))
    return 0;
  return 1;
}

unsigned char Get_XGZPC_Value(float *Value)
{
  float XGZPC_Value = 0;
  if(I2C_RecvByteLen(0xda,0x06,I2C_DataBuff,3))
    {
      XGZPC_Value = I2C_DataBuff[0] * 65536 + I2C_DataBuff[1] * 256 + I2C_DataBuff[2];
       XGZPC_Value = (XGZPC_Value / 8388608) * 100;
    
//      if(XGZPC_Value > 8388608)
//        XGZPC_Value = (XGZPC_Value - 16777216) / 8388608 * 8.192; 
//      else
//        XGZPC_Value = XGZPC_Value / 8388608 * 8.192;
      *Value = XGZPC_Value;
      I2C_DataBuff[0] = 0x0a;
      if(!I2C_SendByteLen(0xda,0x30,I2C_DataBuff,1))
         return 0;
      return 1;
    }
    return 0;
}

stickbreaker, 2019-02-07:
This example code is for a bit bang'd i2c interface. The esp32 has builtin hardware to handle I2c.  It would be better to use it.

Do you have a link for the datasheet that contains the I2C commands for your sensors.  I was unable to find one with a google search.  All I found was the hardware spec, no listing of i2c commands or data fields.

If you can show me the command list I can help you communicate with the sensor.

Chuck.

Bryanbaring, 2019-02-07:
i will ask for my dealer sir.. i post it here soon

Bryanbaring, 2019-02-11:
@stickbreaker seems the supplier is unresponsive for my i2c command request.. is there other idea to find a way how to deal with this sensor sir ?

beegee-tokyo, 2019-02-11:
Is this the [datasheet](http://www.cfsensor.com/editor/attached/file/20170511/20170511105137_9843.pdf) ?
Nothing about I2C commands there but did you try to just read 24 bits (resolution mentioned in the datasheet) from the sensors read address?

Bryanbaring, 2019-02-11:
@beegee-tokyo .sir can you help me on how to code this thing sir.. all i have done is fine the adress of the device and the address is "6D".

stickbreaker, 2019-02-11:
@beegee-tokyo Yes, shitty datasheet.  the posted bitbang code show a possible configuration sequence and readout.  I thought I would try to reverse engineer to something understandable.

the datasheet show the ID as 0x6C or 0x6D but, no protocol info?  What the F?

Chuck.

stickbreaker, 2019-02-11:
@Bryanbaring Try this test code [xgzp test sketch](https://gist.github.com/stickbreaker/60a7305b9eb8afb6e9748dc1b6d5e566)

Chuck.

Bryanbaring, 2019-02-11:
@stickbreaker This is the result from the serial monitor that showed up.
What should I do next, sir?

 (23,22) (23,25) (23,26) (23,27) (23,32) (23,33)
 (25,00) (25,02) (25,04) (25,05) (25,12) (25,13) (25,14) (25,15) (25,16) (25,17) (25,18) (25,19) (25,21) (25,22) (25,23) (25,26) (25,27) (25,32) (25,33)
 (26,00) (26,02) (26,04) (26,05) (26,12) (26,13) (26,14) (26,15) (26,16) (26,17) (26,18) (26,19) (26,21) (26,22) (26,23) (26,25) (26,27) (26,32) (26,33)
 (27,00) (27,02) (27,04) (27,05) (27,12) (27,13) (27,14) (27,15) (27,16) (27,17) (27,18) (27,19) (27,21) (27,22) (27,23) (27,25) (27,26) (27,32) (27,33)
 (32,00) (32,02) (32,04) (32,05) (32,12) (32,13) (32,14) (32,15) (32,16) (32,17) (32,18) (32,19) (32,21) (32,22) (32,23) (32,25) (32,26) (32,27) (32,33)
 (33,00) (33,02) (33,04) (33,05) (33,12) (33,13) (33,14) (33,15) (33,16) (33,17) (33,18) (33,19) (33,21) (33,22) (33,23) (33,25) (33,26) (33,27) (33,32)
 Scanning I2C Addresses
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. 6D .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
Scan Completed, 1 I2C Devices found.
 Adjust SDA, SCL to match pins, and recompile/download
 Locking up

Bryanbaring, 2019-02-11:
@stickbreaker This is the result from the serial monitor that showed up.
What should I do next, sir?
[image]

stickbreaker, 2019-02-11:
@bryanbaring
`Wire.begin(21,22,100000)` in `setup()`.

After that, if my test code doesn't work, then it is a protocol problem. (Documentation needed)

Chuck.

Bryanbaring, 2019-02-11:
@stickbreaker 
huhuh the same result.. 
Reading from 0xA5 failed =(timeout)
Init of sensor at 06D Failed

stickbreaker, 2019-02-11:
@Bryanbaring I2c is working, but we have a failure to communicate with sensor, so that is as far as I can go without additional protocol information from sensor manufacturer

Chuck..

Bryanbaring, 2019-02-11:
this is the ic that they used for this sensor

[CFS 2300+Datasheet+Rev1.9_2018.pdf](https://github.com/espressif/arduino-esp32/files/2851890/CFS.2300%2BDatasheet%2BRev1.9_2018.pdf)

stickbreaker, 2019-02-11:
@Bryanbaring That datasheet makes more sense.  I'll look it over and see if I can figure out an example to communicate with the sensor.  It will take a few days.  Probably by next week end (feb 16).

Chuck.

Bryanbaring, 2019-02-11:
@stickbreaker
ok sir ill wait . thank you very much. .

stickbreaker, 2019-02-11:
@Bryanbaring ReWrote test, try it [xgzp.ino](https://gist.github.com/stickbreaker/60a7305b9eb8afb6e9748dc1b6d5e566)

Chuck.

Bryanbaring, 2019-02-12:
Good day, sir.
It seems that the code is working as desired. I assumed that the unit is KPA. Is it possible that the code could be compressed? 

Thank you, sir.

Best regards,
Bryan Baring

[image]

stickbreaker, 2019-02-12:
@Bryanbaring Try this [XGZP library](https://github.com/stickbreaker/XGZP_6867)

Chuck.

Bryanbaring, 2019-02-13:
@stickbreaker 
this is the result after uploading the sample sketch sir. itried both the find and read 
(    4865ms) I2C bus devices found on (SDA=21, SCL=22)
Scanning I2C Addresses
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. 6D .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
Scan Completed, 1 I2C Devices found.
 Sensor Found, use:

 Wire.begin(21,22,100000);

 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting
 problem reading sensor, Re-Initting

stickbreaker, 2019-02-13:
@Bryanbaring Well, I don't have one of those sensors, to test so I am just guessing.   

If you send me one I'll be able to make something that works.  Unless you are willing to send me one I'm done playing.

The problem you are having is just protocol communication issues.   You should be able to build something from the working parts I have already made.

Chuck.

Bryanbaring, 2019-02-13:
i see @stickbreaker 
iguess ill be doing my part here.. 
thank you very much for your great effort sir , your one of a kind.
what your malling address sir, want to give some of my sensor sir.. 
this might be my simple payback for what youve done to make this project work..

stickbreaker, 2019-02-13:
@Bryanbaring Jump over to [glitter](https://gitter.im/espressif/arduino-esp32?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge) and we'll open a private change, where I will exchange info.

Chuck.

closed 2019-08-15 by stale[bot]

j-u-h, 2021-11-30:
> this is the ic that they used for this sensor
> 
> [CFS 2300+Datasheet+Rev1.9_2018.pdf](https://github.com/espressif/arduino-esp32/files/2851890/CFS.2300%2BDatasheet%2BRev1.9_2018.pdf)

Hi @Bryanbaring ,

I  hope you're still monitoring this, as here is the only source for the datasheet to the CFS2300 chip used on these modules I could find. In fact, without this thread, I still wouldn't know that this chip is used on the XGZP6867 and XGZP6847 modules, as mine are all either cryptically or not labelled at all.

I am currently doing some in depth tests with them, with the aim of writing a more complete library than @stickbreaker's very helpful early effort.

At the moment, I am particularly interested in the OTP calibration data. The datasheet mentions "contacting Novosense customer support for coefficient calculating kits" on p.12. However they haven't answered my requests.

So I am wondering if you could somehow help in providing one of these "kits".

Thank you
Jan

Bryanbaring, 2021-12-02:
Hello jan ,

It's been a while since I used the sensor . I forgot the details. Maybe I can give you the email add of there support tech . So you can directly shoot and email to them.
sales3@cfsensor.com

Thanks.

Regards, 
Bryan

j-u-h, 2021-12-02:
Great, thank you Bryan! I will try that.
Best
Jan

gnieddu90, 2021-12-07:
Good Morning,
I don't understand what is the measure unit, this is my output, in ambiental conditions:
temp = 0
 Value =180.52
temp = 0
 Value =180.52
temp = 0
 Value =180.53
temp = 0
 Value =180.53
temp = 0
 Value =180.53
temp = 0
 Value =180.52
can someone help me?

j-u-h, 2021-12-07:
> Good Morning, I don't understand what is the measure unit, this is my output, in ambiental conditions: temp = 0 Value =180.52

Not sure what code gave you this output. The raw temperature data from the sensor is a signed 16 bit integer. Divide it by 256 to get degrees Celsius (see p5 of the above datasheet).

BTW: cfsensor updated their module datasheets, they now include most of the content of the above "confidential" datasheet and some additional sample code: http://www.cfsensor.com/zcxzinfo/14.html  It seems, though, that the xgzp6867 is obsolete by now.

gnieddu90, 2021-12-07:
> Not sure what code gave you this output. The raw temperature data from the sensor is a signed 16 bit integer. Divide it by 256 to get degrees Celsius (see p5 of the above datasheet).
>
> BTW: cfsensor [...]

The code come from here
[image]

https://gist.github.com/stickbreaker/60a7305b9eb8afb6e9748dc1b6d5e566

Is in the same issue, but I don't understan which value of pressure it is