espressif/arduino-esp32 #2433: Core 1 register dump, Rebooting after using Wifi and OLED I2C

Yeoboseo, 2019-02-07:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
I'm trying to use wifi and I2C OLED realtime, my target actually to use bluetooth too but my priority is wifi. Plz help me

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include "ThingSpeak.h"
#include "secrets.h"  //Containt my wifi pass and ssid, thinspeak id and key. Start with "SECRET_ .."
#include <WiFi.h>
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels
#define OLED_SDA 17
#define OLED_SCL 16
// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define OLED_RESET     4 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

int keyIndex = 0;            // your network key Index number (needed only for WEP)
WiFiClient  client;
unsigned long myChannelNumber = SECRET_CH_ID; // My ID to connet to server
const char * myWriteAPIKey = SECRET_WRITE_APIKEY; // My secret key to upload it
int number = 0;

void setup() {
  Wire.begin(OLED_SDA,OLED_SCL);
  Serial.begin(512000); 
  WiFi.mode(WIFI_STA);   
  ThingSpeak.begin(client);  // Initialize ThingSpeak
  display.display();
}

void loop() {
  display.clearDisplay();
  display.setCursor(0,0);
  display.print("Data:");
  display.setCursor(25,0);
  display.display();
  display.print(number);
  
  // Connect or reconnect to WiFi
  if(WiFi.status() != WL_CONNECTED){
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(SECRET_SSID);
    while(WiFi.status() != WL_CONNECTED){
      WiFi.begin(SECRET_SSID,SECRET_PASS); 
//      Serial.print(".");
      delay(5000);     
    } 
    display.setCursor(0,10);
    display.print("WIFI Connected");
    Serial.println("\nConnected.");
  }
  
  // Write to ThingSpeak. There are up to 8 fields in a channel, allowing you to store up to 8 different
  // pieces of information in a channel.  Here, we write to field 1.
  int x = ThingSpeak.writeField(myChannelNumber, 1, number, myWriteAPIKey);
  if(x == 200){
    Serial.println("Channel update successful.");
  }
  else{
    Serial.println("Problem updating channel. HTTP error code " + String(x));
  }
  
  // Testing the random value
  number=random(60,100);
  
  delay(2000); // Wait 20 seconds to update the channel again
}
```

### Debug Messages:
```
Core 1 register dump:
PC      : 0x400d407c  PS      : 0x00060630  A0      : 0x800d1d74  A1      : 0x3ffb1f60  
A2      : 0x3ffc0b44  A3      : 0x00000000  A4      : 0x00000400  A5      : 0x00000001  
A6      : 0x0000001f  A7      : 0x00000000  A8      : 0x800d4055  A9      : 0x3ffb1f40  
A10     : 0x3ffc0e4c  A11     : 0x00000040  A12     : 0x401433ec  A13     : 0x00000006  
A14     : 0x0000001f  A15     : 0x401433ec  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d407c:0x3ffb1f60 0x400d1d71:0x3ffb1f80 0x400d60c7:0x3ffb1fb0 0x4008f035:0x3ffb1fd0

Rebooting...
```

stickbreaker, 2019-02-07:
@Yeoboseo I think you need to call the init routine for the display
```
void setup(){
//
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // initialize with the I2C addr 0x3D (for the 128x64)
//
```

Chuck.

Yeoboseo, 2019-02-10:
I already tried it but it wont work, it just keep using the default SDA and SCL pins

stickbreaker, 2019-02-10:
@Yeoboseo run this [I2C scan sketch](https://github.com/stickbreaker/arduino-esp32/tree/master/libraries/Wire/examples/i2c_scan).  See if the OLED is visible to the ESP32.

SSD1306 have the possibility of two addresses 0x3C and 0x3D.  It could be wired different than standard.

Chuck.

Yeoboseo, 2019-02-21:
Yes i tried ti get addresses and example of 2 libraries, it works but when wifi turn on the oled get noise

closed 2019-08-15 by stale[bot]