espressif/arduino-esp32 #2435: upload file size undefined?

1TekPro, 2019-02-07:
`/*
      Call this to check the space needed for the update
      Will return false if there is not enough space
    */
    bool begin(size_t size=UPDATE_SIZE_UNKNOWN, int command = U_FLASH, int ledPin = -1, uint8_t ledOn = LOW);`

This does not "check the space needed". Am I missing something or is this actually incomplete and uses max size EVERY time?

lbernstone, 2019-02-08:
This is just a header file.  The C [code](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Updater.cpp#L151) for begin() checks the partition size.

1TekPro, 2019-02-08:
Understood. But still does not do what it says in the comments. At least the comment can be corrected if nothing else.

everslick, 2019-02-13:
@1TekPro the size of the binary for the update has to be passed in by the caller of Updater::begin(); If it is left out, size will default to UPDATE_SIZE_UNKNOWN and no check will be performed. Works as advertized I'd say. no?

1TekPro, 2019-02-13:
@everslick If the size is known in advance, perhaps. This complicates uploads as it is hard coded. If the file size changes with each update, it will break. The way I have suggested fixes this issue when the exact file size is needed for example, when using a progress bar.

everslick, 2019-02-13:
I don't really get what you are trying to solve here. Updater is generic for all kinds of updates, it does not care where the update is coming from (could be a file on an SD card, where the size is known very easily, or the WEB server sends it with CONTENT_LENGTH header). So we call begin() with the correct size, if we know it, or without the size and perform the update hoping it will fit. 

Maybe you should provide a PR with you concrete suggestion and we can discuss that instead?

1TekPro, 2019-02-13:
I am a bit new to github, I assume pr is a pull request?
Using the example of a progress bar, if we know the size of each upload will be the same, we can hard code it. If we dont, the progress bar will only reach around 25% (in my case), because it is using UNKNOWN size.
This is not a bug/issue with the library, but a suggestion for more universal applicability.

balvant813, 2019-07-04:
I get error while compiling in Visual code using arduino ext). How to get around it? `if (!Update.begin(UPDATE_SIZE_UNKNOWN)) { //start with max available size
                Update.printError(Serial);` Can replace UPDATE_SIZE_UNKNOWN with a constant, if yes, what can be reasonable value for WeMos d1 mini etc?

atanisoft, 2019-07-04:
@balvant813 remove the param, it is not needed.

closed 2019-09-16 by stale[bot]