espressif/arduino-esp32 #2437: HTTP Client GET Request doesn't work

GeorgeFlorian, 2019-02-08:
Hardware:

Board: ESP32 DEVKIT1

Core Installation version: 1.0.1-git this one

IDE name: Arduino IDE

Flash Frequency: 80Mhz

PSRAM enabled: no

Upload Speed: 115200

Computer OS: Linux Mint 19.1 Mate

### Description:

I'm trying to integrate make a simple GET Request inside a simple sketch containing [Hieremon's AutoConnect](https://github.com/Hieromon/AutoConnect)

### Sketch: 
 
```
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <HTTPClient.h>

WebServer Server;
AutoConnect Portal(Server);

void rootPage() {
    char content[] = "Hello, world !";
    Server.send(200, "text/plain", content);
  }

void setup () {
    delay(1000);
    Serial.begin(115200);
    Serial.println();
  
    Server.on("/", rootPage);
    if (Portal.begin()) {
        Serial.println("WiFi Connected: " + WiFi.localIP().toString());
      }
  }

void loop() {
    
    HTTPClient http;
    
    http.begin("http://jsonplaceholder.typicode.com/comments?id=10"); //Specify the URL
    int httpCode = http.GET();
  
    if(httpCode > 0) {
        String payload = http.getString();
        Serial.println(httpCode);
        Serial.println(payload);
        
      } else {
          Serial.println("Error on HTTP request");
        }
     http.end();

     Portal.handleClient();     

     delay(10000);
  }
```

### Debug Messages:
```
14:00:02.188 -> ets Jun  8 2016 00:22:57
14:00:02.188 -> 
14:00:02.188 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:00:02.188 -> configsip: 0, SPIWP:0xee
14:00:02.188 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:00:02.188 -> mode:DIO, clock div:1
14:00:02.188 -> load:0x3fff0018,len:4
14:00:02.188 -> load:0x3fff001c,len:1100
14:00:02.188 -> load:0x40078000,len:10088
14:00:02.188 -> load:0x40080400,len:6380
14:00:02.188 -> entry 0x400806a4
14:00:03.505 -> 
14:00:03.505 -> [W][WiFiGeneric.cpp:570] setSleep(): STA has not been started
14:00:03.704 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
14:00:03.704 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
14:00:03.803 -> [AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0) [AC] WiFi.begin()
14:00:03.803 -> [AC] Connecting...[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
14:00:04.699 -> .[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
14:00:04.732 -> [D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.1.101, MASK: 255.255.255.0, GW: 192.168.1.1
14:00:04.997 -> [AC] established IP:192.168.1.101
14:00:04.997 -> [AC] http server started
14:00:05.031 -> WiFi Connected: 192.168.1.101
14:00:05.031 -> [D][HTTPClient.cpp:265] beginInternal(): host: jsonplaceholder.typicode.com port: 80 url: /comments?id=10
14:00:05.064 -> [D][HTTPClient.cpp:977] connect():  connected to jsonplaceholder.typicode.com:80
14:00:05.130 -> [D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
14:00:05.130 -> [D][HTTPClient.cpp:1105] handleHeaderResponse(): size: 322
14:00:05.130 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
14:00:05.163 -> [D][HTTPClient.cpp:1239] writeToStreamDataBlock(): connection closed or file end (written: 322).
14:00:05.163 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:05.163 -> 
14:00:05.163 ->  [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:05.163 -> 
14:00:15.146 -> [D][HTTPClient.cpp:265] beginInternal(): host: jsonplaceholder.typicode.com port: 80 url: /comments?id=10
14:00:15.179 -> [D][HTTPClient.cpp:977] connect():  connected to jsonplaceholder.typicode.com:80
14:00:15.312 -> [D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
14:00:15.312 -> [D][HTTPClient.cpp:1105] handleHeaderResponse(): size: 322
14:00:15.312 -> [D][WiFiClient.cpp:463] connected(): Disconnected: RES: 0, ERR: 128
14:00:15.312 -> [D][HTTPClient.cpp:1239] writeToStreamDataBlock(): connection closed or file end (written: 322).
14:00:15.345 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:15.345 -> 
14:00:15.345 -> is nobis reprehenderit ipsum amet nulla\nquia quas dolores velit et non\naut quia necessitatibus\nnostrum quaerat nulla et accusamus nisi facilis"
14:00:15.345 ->   }
14:00:15.345 -> ]
14:00:15.345 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:15.345 -> 

And so on. 
```

lbernstone, 2019-02-08:
I have no comment on your 3rd party issues, but the HTTPClient example works for me with your URL.  Maybe you are timing out?

GeorgeFlorian, 2019-02-11:
@lbernstone I don't think I am timing out. 

I've tried with this [link](http://webhook.site/5bcc1208-857c-4dd0-8d39-e4c5f87f32b6) and I got this output:

```
14:00:17.504 -> ets Jun  8 2016 00:22:57
14:00:17.536 -> 
14:00:17.536 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:00:17.536 -> configsip: 0, SPIWP:0xee
14:00:17.536 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:00:17.536 -> mode:DIO, clock div:1
14:00:17.536 -> load:0x3fff0018,len:4
14:00:17.536 -> load:0x3fff001c,len:1100
14:00:17.536 -> load:0x40078000,len:10088
14:00:17.536 -> load:0x40080400,len:6380
14:00:17.536 -> entry 0x400806a4
14:00:21.982 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
14:00:21.982 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
14:00:22.878 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
14:00:22.878 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
14:00:22.878 -> [D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.1.102, MASK: 255.255.255.0, GW: 192.168.1.1
14:00:22.977 -> Connecting to WiFi..
14:00:22.977 -> Connected to the WiFi network
14:00:22.977 -> [D][HTTPClient.cpp:265] beginInternal(): host: webhook.site port: 80 url: /5bcc1208-857c-4dd0-8d39-e4c5f87f32b6
14:00:41.286 -> [E][WiFiClient.cpp:220] connect(): lwip_connect_r: 113
14:00:41.286 -> [D][HTTPClient.cpp:970] connect(): failed connect to webhook.site:80
14:00:41.319 -> [W][HTTPClient.cpp:1262] returnError(): error(-1): connection refused
14:00:41.319 -> .[D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:41.319 -> 
14:00:51.305 -> [D][HTTPClient.cpp:265] beginInternal(): host: webhook.site port: 80 url: /5bcc1208-857c-4dd0-8d39-e4c5f87f32b6
14:00:57.344 -> [D][HTTPClient.cpp:977] connect():  connected to webhook.site:80
14:00:57.444 -> [D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
14:00:57.444 -> [D][HTTPClient.cpp:1109] handleHeaderResponse(): Transfer-Encoding: chunked
14:00:57.444 -> [D][HTTPClient.cpp:757] writeToStream():  read chunk len: 23
14:00:57.477 -> [D][HTTPClient.cpp:1239] writeToStreamDataBlock(): connection closed or file end (written: 23).
14:00:57.477 -> [D][HTTPClient.cpp:757] writeToStream():  read chunk len: 0
14:00:57.477 -> [D][HTTPClient.cpp:349] disconnect(): still data in buffer (2), clean up.
14:00:57.477 -> 
14:00:57.477 -> [D][HTTPClient.cpp:358] disconnect(): tcp stop
14:00:57.477 -> 
14:00:57.477 -> Trying to make it work.
14:00:57.477 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:00:57.612 -> 
14:01:07.495 -> [D][HTTPClient.cpp:265] beginInternal(): host: webhook.site port: 80 url: /5bcc1208-857c-4dd0-8d39-e4c5f87f32b6
14:01:07.528 -> [D][HTTPClient.cpp:977] connect():  connected to webhook.site:80
14:01:07.694 -> [D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
14:01:07.694 -> [D][HTTPClient.cpp:1109] handleHeaderResponse(): Transfer-Encoding: chunked
14:01:07.694 -> [D][HTTPClient.cpp:757] writeToStream():  read chunk len: 23
14:01:07.694 -> [D][HTTPClient.cpp:1239] writeToStreamDataBlock(): connection closed or file end (written: 23).
14:01:07.694 -> [D][HTTPClient.cpp:757] writeToStream():  read chunk len: 0
14:01:07.727 -> [D][HTTPClient.cpp:349] disconnect(): still data in buffer (2), clean up.
14:01:07.727 -> 
14:01:07.727 -> [D][HTTPClient.cpp:358] disconnect(): tcp stop
14:01:07.727 -> 
14:01:07.727 -> Trying to make it work.
14:01:07.727 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:01:07.727 -> 
14:01:38.212 -> [D][HTTPClient.cpp:265] beginInternal(): host: webhook.site port: 80 url: /5bcc1208-857c-4dd0-8d39-e4c5f87f32b6
14:01:56.455 -> [E][WiFiClient.cpp:220] connect(): lwip_connect_r: 113
14:01:56.455 -> [D][HTTPClient.cpp:970] connect(): failed connect to webhook.site:80
14:01:56.455 -> [W][HTTPClient.cpp:1262] returnError(): error(-1): connection refused
14:01:56.488 -> f[D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:01:56.488 -> 
14:02:06.472 -> [D][HTTPClient.cpp:265] beginInternal(): host: webhook.site port: 80 url: /5bcc1208-857c-4dd0-8d39-e4c5f87f32b6
14:02:24.984 -> [E][WiFiClient.cpp:220] connect(): lwip_connect_r: 113
14:02:24.984 -> [D][HTTPClient.cpp:970] connect(): failed connect to webhook.site:80
14:02:24.984 -> [W][HTTPClient.cpp:1262] returnError(): error(-1): connection refused
14:02:24.984 -> [D][HTTPClient.cpp:369] disconnect(): tcp is closed
14:02:24.984 -> 

```

As you can see it doesn't work all the time. The URL is from Webhook. It should work. That's why Webhook exists.

```
14:00:41.286 -> [D][HTTPClient.cpp:970] connect(): failed connect to webhook.site:80
14:00:41.319 -> [W][HTTPClient.cpp:1262] returnError(): error(-1): connection refused
```

I still don't know if this works reliably. It still doesn't work with the first URL.

GeorgeFlorian, 2019-02-21:
I've closed this because I've started using PlatformIO IDE and it works just fine.

closed 2019-02-21 by GeorgeFlorian

ashrafkamel5, 2019-10-19:
it is clearly that it was a bad version of the http library that platformIO fixed 
can you tell me which library version that worked fine with you?
i have to reconnect to the wifi every time i want to send a http request and it takes lots of time