espressif/arduino-esp32 #2441: Just want to verify this does NOT work with IOS

Sparkinman, 2019-02-10:
As far as I can tell, this will not work with IOS devices, can anyone please confirm for sure?

lbernstone, 2019-02-10:
arduino-esp32 is quite specifically for programming the [ESP32 SoC](https://www.espressif.com/en/products/hardware/esp32/overview).  [Arduino](https://www.arduino.cc) is a platform independent programming IDE.  It does not run on IOS, because if you can afford a $1000 phone, you probably have a real computer.

Sparkinman, 2019-02-10:
Not exactly what I meant!  As near as I can tell, there is no IOS support for serial over bluetooth communication (SPP), rendering this library unusable on IOS devices...correct?

lbernstone, 2019-02-10:
Sorry, my mind reading doesn't work so well on Sunday. https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

Sparkinman, 2019-02-10:
I see that on the front page, it SAYS that it works with IOS, but using the example sketch, and the recommended app on IOS, the device doesn't even show up in a BT scan (however from Windows, Android, and Mac) it does, and works fine from them.  Am I doing something wrong, or is it simply IOS being a problem, and help would be appreciated!

Arduingo, 2019-04-07:
One reason is because the suggested iOS App (https://github.com/hoiberg/HM10-BluetoothSerial-iOS) is scanning for a different service a Characteristic IDs

    /// UUID of the service to look for.
    var serviceUUID = CBUUID(string: "FFE0")
    
    /// UUID of the characteristic to look for.
    var characteristicUUID = CBUUID(string: "FFE1")

chegewara, 2019-04-07:
> One reason is because the suggested iOS App (https://github.com/hoiberg/HM10-BluetoothSerial-iOS) is scanning for a different service a Characteristic IDs
>
> ```
> /// UUID of the service to look for. [...]

This is BLE app, BluetoothSerial is classic BT protocol.

Arduingo, 2019-04-07:
> This is BLE app, BluetoothSerial is classic BT protocol.

The HM10 is a BLE module that can be used as an UART, I have used the example with own "terminal" writing and reading data using BLE, now BLE does not support native UART so each vendor should implement their own "Simulated" UART. The HM10 implemented this, but the sketch espressif is giving us is not compatible. In other words the suggested app in IOS is useless.

closed 2019-08-15 by stale[bot]