espressif/arduino-esp32 #2442: LEDC won't work with freq over 78.125 Khz

wgaylord, 2019-02-10:
### Hardware:
Board:							ESP-WROOM-32
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200?
Computer OS:                                        Windows 10

### Description:
LEDC won't output a freq anywhere close to the maximum freq as described in the docs.

Seems to specifically be a problem with ledcWriteTone 

### Sketch:  ```
void setup() {
  Serial.begin(115200);

     Serial.printf("frequency: %f\n", ledcSetup(1,7000000,1)/1000);
     ledcAttachPin(22,1);
       
 Serial.printf("frequency: %fKhz\n", ledcWriteTone(1,7000000)/1000);
 
}

void loop() {vTaskDelete(NULL);}
```

### Debug Messages:
```
None
```

###Aditional Information
Using 
```
#include "driver/ledc.h"

#define LEDC_0_PIN 22

void setup() {
  Serial.begin(115200);
  
  ledc_timer_config_t ledc_timer;
  ledc_timer.speed_mode = LEDC_HIGH_SPEED_MODE;           // timer mode
  ledc_timer.duty_resolution = LEDC_TIMER_3_BIT; // resolution of PWM duty
  ledc_timer.timer_num = LEDC_TIMER_0;            // timer index
  ledc_timer.freq_hz = 7000000;                      // frequency of PWM signal
  // Set configuration of timer0 for high speed channels
  esp_err_t result = ledc_timer_config(&ledc_timer);
  if (result == ESP_OK)
     Serial.printf("frequency: %d", ledc_get_freq(LEDC_HIGH_SPEED_MODE, LEDC_TIMER_0));
  ledc_channel_config_t ledc_channel = {
          .gpio_num   = LEDC_0_PIN,
          .speed_mode = LEDC_HIGH_SPEED_MODE,
          .channel    = LEDC_CHANNEL_0,
          .intr_type  = LEDC_INTR_DISABLE,
          .timer_sel  = LEDC_TIMER_0,
          .duty       = 4,
          .hpoint     = 0
  };
  // Set LED Controller with previously prepared configuration
  ledc_channel_config(&ledc_channel);
}

void loop() {vTaskDelete(NULL);}

```
Does output a freq close to 7 Mhz (Exactly 7.013698 Mhz)

lbernstone, 2019-02-10:
This is an [esp-idf issue](https://github.com/espressif/esp-idf/issues), but the lower the resolution, the higher frequency you will be able to achieve.

wgaylord, 2019-02-10:
So the arduino functions that don't work is an esp-idf issue?

And I have a single bit for resolution. 

I can get 40Mhz output from the IDF functions but not from the Arduino Wrapper functions.

wgaylord, 2019-02-10:
More information using ledcSetup I can get up to the max 40Mhz but not using the function meant for just updating the freq.

lbernstone, 2019-02-10:
The [arduino ledc functions](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.h) are different than the [esp-idf functions](https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/ledc.h) you are using.

wgaylord, 2019-02-10:
ledcWriteTone and ledcSetup are not the arduino ledc functions?

wgaylord, 2019-02-10:
I provided addition information at the end to say that the IDF functions work and that the arudino ones do not.

wgaylord, 2019-02-11:
So any idea on why ledcWriteTone will not set higher then 78.125Khz?

wgaylord, 2019-02-11:
Seems I figured it out ledcWirteTone uses https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c#L240
It doesn't just adjust the existing timer.

closed 2019-02-11 by wgaylord