espressif/arduino-esp32 #2443: Esp32 4x4 keypad

palsayantan, 2019-02-11:
I was trying to connect a 4x4 alphanumeric keypad with the ESP32 based NodeMCU board. I want to connect them to the ADC pins of nodeMCU. So I modified the example customkeypad code and assigned the row and column pins to 36,39,34,35 and 32,33,25,26 respectively. But that didn't work and I get no response. Please help.

GeorgeFlorian, 2019-11-19:
Hello ! I would like to try something similar. Did you manage to make it work ?

palsayantan, 2019-11-19:
Yah it's working properly

closed 2019-11-19 by palsayantan

GeorgeFlorian, 2019-11-19:
> Yah it's working properly

And how did you do it ?

palsayantan, 2019-11-19:
Actually you can't use certain pins on the esp32 with matrix keypad. For me it won't work on the pin 34,35,36 and 39. So I changed the pin numbers in the Code and it worked. 
You can check out my SMART VENDING MACHINE repo for reference.

atanisoft, 2019-11-19:
Pins 34-39 are input only and can be used for matrix inputs if the library is configured correctly.

GeorgeFlorian, 2019-11-19:
> Pins 34-39 are input only and can be used for matrix inputs if the library is configured correctly.

What library ? How can I use a keypad ? I don't suppose that I'll just buy one and connect it to the ESP and it will work like magic.

atanisoft, 2019-11-19:
> What library ? How can I use a keypad ? I don't suppose that I'll just buy one and connect it to the ESP and it will work like magic.

A very quick search here on GitHub resulted in this lib https://github.com/Nullkraft/Keypad which appears that it should work with the esp32. The rows pins are used as INPUT and cols are OUTPUT.

GeorgeFlorian, 2019-11-20:
> A very quick search here on GitHub resulted in this lib https://github.com/Nullkraft/Keypad which appears that it should work with the esp32. The rows pins are used as INPUT and cols are OUTPUT.

I guess I don't have to use 1 pin per key, correct ?

I actually need a bigger keypad, something more like a keyboard. Something that has all the alphanumerical keys, an enter and a backspace.

Can this be done in ESP32 ? Or should I move to a Raspberry Pi Zero ?

I need something like this:
[image]

palsayantan, 2019-11-20:
You can visit electronoobs channel on YouTube. He worked on a similar project