espressif/arduino-esp32 #2444: i2c problem with lcd display 16x2

atlask1, 2019-02-11:
### Hardware:
Board:							ESP32 DevKit v1
Core Installation version:			1.0.1-git
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Ubuntu

### Description:
With esp32 ,after 2 month of work without problem, my lcd 16x2 with PCF8574 (external component) have issue that I don't understand. The inizialization don't start and I receive error on i2c.
I've seen similar problem in other issue , maybe solved with 'setTimeout()' of Wire library, but I'm not using Wire.h directly , I'm using a library for hd44780, and the problem readed was   #462
In particular I use hd44780ioClass/hd44780_I2Cexp.h
[hd44780.zip](https://github.com/espressif/arduino-esp32/files/2851379/hd44780.zip)

When I debug the sketch I receive this : 

[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0x54, end=0x86, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdda4
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbbec4
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbdaf8 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbbf24
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffbdad4
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0xa5, end=0xd7, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdda4
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbbec4
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbdaf8 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbbf24
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffbdad4
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[D][esp32-hal-i2c.c:1297] i2cProcQueue():  Busy Timeout start=0xf2, end=0x124, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbdda4
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbbec4
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbdaf8 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbbf24
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffbdad4
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[V][esp32-hal-i2c.c:1442] i2cInit(): num=0 sda=21 scl=22 freq=0
[V][esp32-hal-i2c.c:1643] i2cSetFrequency(): Fifo threshold=3
[D][esp32-hal-i2c.c:1306] i2cProcQueue():  Gross Timeout Dead start=0x143, end=0x175, =50, max=50 error=0

after a lot of reboot , finally lcd initialize, but more slow than before , and with garbage(sometimes) or error of setCursor() or no lcd.clear(), and after another flash I receive the same error.
Now is blocked with this error.
What can I do?
thanks

atlask1, 2019-02-11:
Sorry I see a similar problem on #834, a closed issue, not #462

stickbreaker, 2019-02-11:
@atlask1 Do you have a schematic of the connections between the LCD and the IO Expander?  If so, post it. 

The Errors you are seeing indicate something is holding the bus
>[D][esp32-hal-i2c.c:1297] i2cProcQueue(): Busy Timeout start=0x54, end=0x86, =50, max=50 error=0

To my knowledge, the PCF8574 has no ability to stall the bus by clock stretching.  So, something else is going on.

Those issues you reference are not applicable to the current I2C driver.  You are seeing something else 

Chuck.

atlask1, 2019-02-12:
Hi @stickbreaker , 
here the schematic, but is strange that work for 2 month and now I have problem, without changing sketch or  library, from one day to another lcd don't work. 
Now is ok but more slow, but I can't consider it stable.
what you can see is a homemade board with lcd and pcf8574 , the i2c go to D21 and D22 of esp32 devkit (i2c pins).
[image]

thanks

atlask1, 2019-02-12:
You said ‘To my knowledge, the PCF8574 has no ability to stall the bus by clock stretching’ 
So, for you,

closed 2019-04-18 by atlask1