espressif/arduino-esp32 #2446: CONFIG_APP_ROLLBACK_ENABLE not enabled?

vicatcu, 2019-02-11:
I'm using v1.0.1 of the Boards Manager package. It seems to me CONFIG_APP_ROLLBACK_ENABLE is _not enabled_ for OTA in the sdkconfig for the Arduino core. Is there any way to _make_ it be enabled? Otherwise it seems like a bad idea for the default to be not enabled no?

me-no-dev, 2019-02-14:
Support is coming to master :) added here: https://github.com/espressif/arduino-esp32/commit/d3534b3dda42f15e1d5677711ed133ef82c76a23
Thanks for the suggestion!

closed 2019-02-14 by me-no-dev

MarkLogiCom, 2019-02-15:
Hi me-no-dev 
Don't know if this is related but I just fetched latest (Feb 15) and it broke my build...
"----------------------------------------------
C:\SPB_Data\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c: In function 'initArduino':
C:\SPB_Data\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-misc.c:164:5: error: unknown type name 'esp_ota_img_states_t'
--------------------------------------------------------
Did I miss step? Deleted CONFIG_APP_ROLLBACK_ENABLE from my sdkcongig.h and all is good again.
esp_ota_img_states_t ota_state;
^

me-no-dev, 2019-02-15:
@MarkLogiCom using as IDF component?

MarkLogiCom, 2019-02-15:
Sorry should have left you my setup.
ESP32 Dev Kit C, Arduino framework, just building the Basic OT example and discovered no longer built - but hadn’t sync’d to Github repo for a while.
Was building OK prior to latest fetch, but failed afterwards.
Did a diff against my last known save(zip) of the framework and found CONFIG_APP_ROLLBACK_ENABLE was added to sdkconfig.h. I removed this and my build was now successful.

Hope that helps - I am a noob to PlatformIO/Arduino so forgive me for not knowing what an “IDF component” is.
Regards
Mark

me-no-dev, 2019-02-15:
hmmm ... i verified with the same example. You can see the definitions here: https://github.com/espressif/arduino-esp32/search?q=esp_ota_img_states_t&unscoped_q=esp_ota_img_states_t
Maybe delete the repo and clone again? It is possible that something went wrong with pulling the latest code :)

MarkLogiCom, 2019-02-15:
You could well be right, I have been doing lot of different test code/projects and probably mangled my repo.
When I get a chance, I will clean up and re-clone and let you know.
I would leave the original issue closed.
Thanks again for your time