espressif/arduino-esp32 #2449: [BLE] Any plans for BLE Mesh support on the arduino side?

marquesn, 2019-02-12:
Pretty much as the title says - if there are any future plans of bringing the BLE mesh capability over from the ESP-IDF to the arduino side?

chegewara, 2019-02-12:
It would be nice, but ble mesh is not supported yet in esp-idf. It is only supported in separate branch.
Yesterday ive been looking at ble mesh and how i could wrap it with c++, but it will take some time, especially i am quite busy. 

Maybe someone else could do it.

chanjackie, 2019-07-16:
Is there any update on BLE mesh support with Arduino? Or will it have to wait until Espressif releases a stable release with BLE mesh?

tedder, 2019-08-15:
#1645 went stale, was the same thing.

atanisoft, 2019-08-15:
@tedder #1645 was for WiFi mesh and not BLE mesh so not the same thing. BLE Mesh does not appear to be available in ESP-IDF in a stable release yet. Until it shows up in a stable release it won't make it to the arduino-esp32 side except through an external library possibly.

matt-attack, 2019-10-09:
It looks like it has been released now. Is there any chance of this happening? I'd be happy to help if there's something relatively straightforward to do for this.

atanisoft, 2019-10-09:
@matt-attack is it on esp-idf v3.3 or v3.2? If it is only on v4.x beta it is not ready for use as only stable releases are integrated with arduino-esp32.

matt-attack, 2019-10-09:
@atanisoft Ah. Didn't realize that there were beta releases, I assumed they were all, well, stable.

atanisoft, 2019-10-09:
@matt-attack in general they are relatively stable but are not considered stable until a final release is made.

closed 2019-12-22 by stale[bot]

TheGreenToaster, 2020-02-11:
Looks like 4.0 was released today.

atanisoft, 2020-02-11:
@TheGreenToaster it will show up soon enough in the v4.0 branch on here

faxik, 2020-03-24:
Still impossible to compile: I've added details in #3846.