espressif/arduino-esp32 #2450: Changing timezone with setenv() causes memory leaks (with workaround)

lszlfrng, 2019-02-12:
### Hardware:
Board: ESP32-DevkitC
Core Installation version:	1.0.1		
IDE name: Platform.io CLI with Eclipse
Flash Frequency: 40Mhz
PSRAM enabled: yes
Upload Speed: 921600
Computer OS: Windows 10

### Description:
In my project I have to change the timezone frequently. For this I use the time.h setenv("TZ","timezone",1) ). 
When the length of the time zone value increases the free heap will be smaller and smaller.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
uint32_t h0 = 0;
uint32_t h = 0;
void setup() {
	Serial.begin(115200);
	log_d("Initial Heap: %lid", ESP.getFreeHeap());
	h0 = ESP.getFreeHeap();
}

void loop() {
	log_d("");
	h = ESP.getFreeHeap();
	setenv("TZ", "GMT1GMT1", 1);
	log_d("Heap GMT1:  lost:% 5li Σlost:% 5li",  h-ESP.getFreeHeap(), h0-ESP.getFreeHeap());
	h = ESP.getFreeHeap();
	setenv("TZ", "GMT-1GMT-1", 1);
	log_d("Heap GMT-1: lost:% 5li Σlost:% 5li",  h-ESP.getFreeHeap(), h0-ESP.getFreeHeap());
}
```

### The result is:

```
[D][main.cpp:46] setup(): Initial Heap: 379364d
[D][main.cpp:53] loop(): Heap GMT2:  lost:  272 Σlost:  272 time:Fri Dec 31 22:00:00 1999
[D][main.cpp:56] loop(): Heap GMT-2: lost:   32 Σlost:  400 time:Sat Jan  1 02:00:00 2000
[D][main.cpp:53] loop(): Heap GMT2:  lost:    0 Σlost:  400 time:Fri Dec 31 22:00:00 1999
[D][main.cpp:56] loop(): Heap GMT-2: lost:   32 Σlost:  432 time:Sat Jan  1 02:00:00 2000
[D][main.cpp:53] loop(): Heap GMT2:  lost:    0 Σlost:  432 time:Fri Dec 31 22:00:00 1999
[D][main.cpp:56] loop(): Heap GMT-2: lost:   32 Σlost:  464 time:Sat Jan  1 02:00:00 2000
[D][main.cpp:53] loop(): Heap GMT2:  lost:    0 Σlost:  464 time:Fri Dec 31 22:00:00 1999
[D][main.cpp:56] loop(): Heap GMT-2: lost:   32 Σlost:  496 time:Sat Jan  1 02:00:00 2000
[D][main.cpp:53] loop(): Heap GMT2:  lost:    0 Σlost:  496 time:Fri Dec 31 22:00:00 1999
[D][main.cpp:56] loop(): Heap GMT-2: lost:   32 Σlost:  528 time:Sat Jan  1 02:00:00 2000
[D][main.cpp:53] loop(): Heap GMT2:  lost:    0 Σlost:  528 time:Fri Dec 31 22:00:00 1999
```

### Workaround 

Keep the "TZ" environment variable length fixed. Example:
```
	setenv("TZ", "GMT2GMT2  ", 1);
....
	setenv("TZ", "GMT-2GMT-2", 1);
```

#### Disadvantage

Not easy to hold the length of the timezone if you have to use the Daylight Saving Time (DST, summer time).
Based on actual environment value you can padding the new value with whitespace.

```
	setenv("TZ", "CET-1CEST,M3.5.0/2,M10.5.0/3", 1);
	setenv("TZ", "GMT0                        ", 1);
```

### Question

Is there any core solution to handle the UTC and local time bidirectional conversation?

atanisoft, 2019-02-12:
This should likely be routed to esp-idf as I don't believe these functions come from the arduino-esp32 side

lszlfrng, 2019-02-12:
Moved to https://github.com/espressif/esp-idf/issues/3046

closed 2019-02-12 by lszlfrng

ghmpi, 2023-12-19:
Hello,

I know this is old.. but it's 2023 and the issue is still there. Took me a while to figure out where the memory leak was from.
Here is a quick and easy fix that worked for me. It pads the TZ variable with spaces. Just use this function to change the TZ variable. Hope it helps someone.

Regards,
-Moses

`
void setTimezone(String timezone){
  int len = timezone.length();
  for(int i = 0; i < (60 - len); i++) { // 12/17/2023 ADDITION, Code to add spaces to TZ (timezone) environment variable due to a memory leak in the way these variables work. This fixes the issue. Before it was causing a memory leak and crashing the ESP every 5-10 hours or so, depending on how many times TZ was changed.
    timezone += ' ';  
  }
  
  setenv("TZ",timezone.c_str(),1);  //  Now adjust the TZ.  Clock settings are adjusted to show the new local time
  tzset();
}
`