espressif/arduino-esp32 #2455: unsigned long millis() timer does not wrap / ignores top 10 bits / causes regression

dirkx, 2019-02-13:
The fix for 2250, commit daea578841059fef36f98d72cef15e22733bdc3e, changed the `unsigned long millis()` into returning `(unsigned long) micros()/1000`.

This potentially seems to have two side effects; firstly it means the top 10 bits of this unsigned long are no longer used (micros() is also an unsigned long; wrapping every 1 hour and 12 minutes - so millis() follows suit in the bottom 21 bits). 

And secondly it breaks the time honoured (and very common code in Arduino libraries) of the type:

     unsigned long last = 0;
     
     void loop() {
         // Use a substract to make sure that this also works when millis() wraps
        // around. See https://playground.arduino.cc/Code/TimingRollover
         if (millis() - last > 5*1000UL) {
               Serial.println("5 second tock");
               last = millis();
        };   
     
Such as used in all sort of TCP clients, SSL session key renewals, MQTT ping/acks, websocket callbacks and so on.

Board:							All ESP
Core Installation version:			master from git as of today.
IDE name:						Arduino IDE, Platformio
Flash Frequency:					All
PSRAM enabled:                                    either
Upload Speed:					115200
Computer OS:                                        OSX

The ideal solution would perhaps be to use a slower timer; or to derive the millis() form micros - but let it keep its own top 10 bit accounting.

dirkx, 2019-02-13:
Pull that introduced this: #2250

stickbreaker, 2019-02-13:
See PR #2438

dirkx, 2019-02-13:
Changing it to 
      return (unsigned long) (micros64() / 1000);
ought to work if that is a decent 64 bit timer.

stickbreaker, 2019-02-13:
@dirkx the problem is that the compiler takes short cuts(optimization).  The result is a 32bit value, one param is a 32 value, so it down converts the other param (64bit value) to a 32bit value(loosing sigs) then does the division.  That pr's (#2438) only change is to force the compiler to do 64bit math before the down conversion to 32bits.

Try that pr's soluction, just change the divisor to `1000ULL`.  Problem's gone.

Chuck.

dirkx, 2019-02-13:
Makes total sense.

dirkx, 2019-02-23:
#2428 solves the issue - and also fixes the MQTT and DHCP hangs.

abqmichael, 2019-07-03:
Has this fix been released? I am still encountering the issue of the value of millis() wrapping around at 2^32/1000 on esp32 version 1.0.2. The bug was quite confounding for me to uncover and I'm sure anybody using any form of event-based programming will run into it quickly. 

Esp32 version 1.0.2 is the most updated version that is available on Arduino.

abqmichael, 2019-07-03:
Oops. I stand corrected. I did a fresh install of the ESP32 board library and it looks like the fix is in version 1.02.

My problem, apparently, was that though Arduino thought it had the newest library, my install was pointing to an old one. So I uninstalled the ESP32 board library and deleted a subdirectory with the driver code from my Arduino directory. Then did a fresh install, re-ran my tests and have broken through the prior barrier.

closed 2019-09-15 by stale[bot]