espressif/arduino-esp32 #2456: Esp32 with Gps module Neo 6m

misha11720, 2019-02-13:
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation version:			 ?1.0.1? 
IDE name:						?Arduino IDE? 
Flash Frequency:					?80Mhz?
PSRAM enabled:                                    ?no? 
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10?

### Description:
I want to interface esp32 with ublox neo 6m. I have found a code mentioned below in which TinyGps++ and SoftwareSerial libraries are used. When I compiled the sketch following error was generated:
 
Arduino: 1.8.8 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, Info"
C:\Users\Bhatti\Documents\Arduino\libraries\espsoftwareserial-master\SoftwareSerial.cpp:27:18: fatal error: gpio.h: No such file or directory

Afterwards, I found a code for serial communication in esp32 using HardwareSerial library. The code is mentioned below. Again I compiled the sketch but failed. This time the error generated is as follows:

C:\Users\Bhatti\AppData\Local\Temp\arduino_cache_717614\core\core_c75e6e14d89bce4be8628b58a96344a5.a(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'

C:\Users\Bhatti\AppData\Local\Temp\arduino_build_332228\sketch\serial_communication.ino.cpp.o:(.bss.Serial2+0x0): first defined here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
//Code for Gps
#include <TinyGPS++.h>
#include <SoftwareSerial.h>

static const int RXPin = 4, TXPin = 3;
static const uint32_t GPSBaud = 115200;

// The TinyGPS++ object
TinyGPSPlus gps;

// The serial connection to the GPS device
SoftwareSerial ss(RXPin, TXPin);

void setup(){
  Serial.begin(9600);
  ss.begin(GPSBaud);
}

void loop(){
  // This sketch displays information every time a new sentence is correctly encoded.
  while (ss.available() > 0){
    gps.encode(ss.read());
    if (gps.location.isUpdated()){
      Serial.print("Latitude= "); 
      Serial.print(gps.location.lat(), 6);
      Serial.print(" Longitude= "); 
      Serial.println(gps.location.lng(), 6);
    }
  }
}
//Code for serial communication after replacing SoftwareSerial with HardwareSerial
#include "HardwareSerial.h"
//HardwareSerial Serial1(1);
HardwareSerial Serial2(2);

#define RXD2 16
#define TXD2 17

void setup() {
  // Note the format for setting a serial port is as follows: Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial.begin(115200);
  //Serial1.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
 Serial.println("Serial Txd is on pin: "+String(TX));
 Serial.println("Serial Rxd is on pin: "+String(RX));
}

void loop() { //Choose Serial1 or Serial2 as required
  while (Serial2.available()) {
   Serial.print(char(Serial1.read()));
  }
}
```

### Debug Messages:
```
//for first code
Arduino: 1.8.8 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, Info"
C:\Users\Bhatti\Documents\Arduino\libraries\espsoftwareserial-master\SoftwareSerial.cpp:27:18: fatal error: gpio.h: No such file or directory
//for second code
C:\Users\Bhatti\AppData\Local\Temp\arduino_cache_717614\core\core_c75e6e14d89bce4be8628b58a96344a5.a(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'

C:\Users\Bhatti\AppData\Local\Temp\arduino_build_332228\sketch\serial_communication.ino.cpp.o:(.bss.Serial2+0x0): first defined here

```

MarkyAD, 2019-02-14:
Dont call the ports Serial1, 2, etc because those names are already defined. Use something like:
```c
HardwareSerial gps_serial(2);

void setup() {
 gps_serial.begin(115200, SERIAL_8N1, RXD2, TXD2);
}
```
Be careful using the symbol "gps" for the port because you've already got a TinyGPS object instance with that name. Also get rid of anything having to do with softwareserial, there is no need for its includes or objects.

misha11720, 2019-02-24:
Thank you so much.

closed 2019-02-24 by misha11720

pono1012, 2020-06-09:
Do you have a finished code, i only get zeros from my gps

sbonaime, 2020-09-09:
@misha11720 I still have the error
` fatal error: gpio.h: No such file or directory
 #include "gpio.h"`

with the following code

``` Arduino
//Change the code below by your sketch
//Code for Gps
#include <TinyGPS++.h>
#include <SoftwareSerial.h>

static const int RXPin = 4, TXPin = 3;
static const uint32_t GPSBaud = 115200;

// The TinyGPS++ object
TinyGPSPlus gps;

#include "HardwareSerial.h"
HardwareSerial gps_serial(2);

#define RXD2 16
#define TXD2 17

void setup() {
  // Note the format for setting a serial port is as follows: Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial.begin(115200);
 
 gps_serial.begin(115200, SERIAL_8N1, RXD2, TXD2);

 Serial.println("Serial Txd is on pin: "+String(TX));
 Serial.println("Serial Rxd is on pin: "+String(RX));
}

void loop() {
  while (gps_serial.available()) {
   Serial.print(char(gps_serial.read()));  // read from gps, write to serial debug port
  }
}
``` 

Can you post your code ? 
Thank you

sbonaime, 2020-09-09:
`#include <SoftwareSerial.h>`
needs to be replaced by
`#include <HardwareSerial.h>`