espressif/arduino-esp32 #2459: Deleting old records from SPIFFS

rado99, 2019-02-14:
Hello, I need something like a round-robin database with a file. I am using SPIFFS.h and FS.h, integrated in the Arduino IDE. So, let's explain my problem:
I save a log that is long around 30 bytes every 5 minutes and the free space of the Flash (4MB) will be full in around a year. In this case, I will need to delete the old records to save the fresh one. I know, that File_Write deletes the whole file before writing and File_append writes on the end of it. 
Thank you in advance.

GeorgeFlorian, 2019-03-13:
Hello !
Did you manage to fix your issue ? If yes, can you please share your resolve ?

Also, how did you manage to write logs to a SPIFFS file ?

**I want to do a similar project**:

> I would like to display log messages to a webpage hosted on the ESP32.
> And I want to do that by writing them first inside a SPIFFS file.
> Let’s say I have a html page events.html and a HTTP GET with a URL inside the void loop(). For example, I would like to check logs for the HTTP GET request. Or let’s say that the ESP crashes out of nowhere and I would like to access some logs and see what went wrong.

rado99, 2019-03-14:
Hello @GeorgeFlorian ,
First, I fixed the problem while I am using more than one log file. Every of them is with fixed size.
Althought, I used this:  https://techtutorialsx.com/2017/12/01/esp32-arduino-asynchronous-http-webserver/. Using this tutorial, you can link directly a file on the SPIFFS to an HTTP request link. You need to have something similar to this:
   ```
server.on("/logs.txt", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/logs.txt", "text/plain");
  });
```
 For other questions, don't hesitate to ask me.

GeorgeFlorian, 2019-03-14:
> Hello @GeorgeFlorian ,
> First, I fixed the problem while I am using more than one log file. Every of them is with fixed size.
> Althought, I used this: [...]

Thank you for your answer !
I'm also using AsyncWebServer to serve webpages.

But how did you redirect the debug messages towards a SPIFFS File ?

closed 2019-08-15 by stale[bot]