espressif/arduino-esp32 #2460: espota.py update routed from different subnet fails

Harvie, 2019-02-14:
Hello, i've found that espota is not working through the routed network. PC has to be in the same network as the ESP. I've already described this problem in following threads, so details are there:

https://github.com/esp8266/Arduino/issues/5341
https://github.com/arduino/Arduino/issues/8196

lbernstone, 2019-02-14:
AFAICT, there is nothing fancy in the communications in espota.  Make sure udp passes both directions.  It is far more likely that it is blocked going from wifi into your main network than the other direction.

closed 2019-08-15 by stale[bot]

Harvie, 2019-09-19:
@lbernstone in fact the traffic passes in both directions to other devices on that wifi. i just think that esp32 code does not send the packets through the default route it gets from DHCP. or something like that...

chimeranzl, 2021-10-31:
Old post I know, however I found a solution to this.  First I created a multicast forwarding policy on my firewall, and then a policy to pass traffic between my LAN and "IOT WiFi" subnets (which gave me the visibility of the ESP8266 IP address under Arduino IDE, Ports).  I had an issue with uploading my sketch across over the firewall though I kept getting "No response from device".  So I edited espota.py and changed the following lines to include a timeout:

WAS this...

  try:
    sock.settimeout(10)
    connection, client_address = sock.accept()
    sock.settimeout(none)
    connection.settimeout(none)
  except Exception:
    logging.error('No response from device')
    sock.close()
    return 1

Changed to this:

  try:
    sock.settimeout(10)
    connection, client_address = sock.accept()
    sock.settimeout(10)
    connection.settimeout(10)
  except Exception:
    logging.error('No response from device')
    sock.close()
    return 1

And now it uploads, however it is very slow (compared to uploading when on the same network) and sometimes errors before completion, so overall I think I'll give up on this and use web updating instead.