espressif/arduino-esp32 #2461: Esp32 Pico Kit & CameraWebServer memory issue

hunterschoening, 2019-02-14:
Hey guys, I'm using the ESP32 Pico Kit V4 and the OV2640 camera module and attempting to upload the CameraWebServer example sketch to my board, but am faced with this memory error upon uploading. As you can see in the image, it says I am using 166% of the program storage space, and I'm not so sure what to do to fix this and get it working. If anyone has any ideas or has resolved this issue any help would be much appreciated 
[image]

lbernstone, 2019-02-14:
Switch to ESP32 dev module and choose huge app partition scheme.

hunterschoening, 2019-02-14:
Thanks Ibernstone,

The flash is now successful, using 69% of the program storage space rather than 166%, but I am now being faced with this error when opening up the serial terminal. Do you have any idea if this is maybe just a wiring issue? I have IO0 connected to ground during the flash period, and then remove it from ground and pressing the enable button, leading to this error. I am attaching a picture of the error as well as the camera pin mapping as it is currently set. Thanks so much for the help so far and thanks in advance for any more help you can provide!
[image]

[image]

closed 2019-08-15 by stale[bot]