espressif/arduino-esp32 #2462: ESP32 & OV2640 Camera not supported issue

hunterschoening, 2019-02-14:
Hey guys,

I am attempting to flash the CameraWebServer example code to the ESP32 Pico Kit V4 using the OV2640 camera module. I've been helped with the memory issue when flashing and can now successfully flash to the board, but am now faced with this issue regarding the camera not being supported. I've attached two images below, one which shows the error, and another picture showing the current pin mapping for the camera to the ESP32 chip. Thanks in advance for any help!

[image]

[image]

me-no-dev, 2019-02-14:
this points to a problem with SIOD and SIOC lines. This is in reality Software I2C. Maybe add pull-ups to those lines?

hunterschoening, 2019-02-14:
Thanks @me-no-dev,

Your answer solved the issue with the "Detected camera not supported" error, but now that it is wired up with some pull-ups on the SIOD and SIOC lines, it is still returning with the "esp_camera_init(): Camera probe failed with error 0x20001" as you can see in the image attached here. Any further ideas on what could resolve this? Thanks again for the help and thanks in advance for any further assistance you can provide!
[image]

me-no-dev, 2019-02-14:
are you sure you got the correct pins? do you have another camera to try?

closed 2019-08-15 by stale[bot]

nbtronix, 2019-09-10:
> Hey guys,
>
> I am attempting to flash the CameraWebServer example code to the ESP32 Pico Kit V4 using the OV2640 camera module. I've been helped with the memory issue when flashing and can now [...]

You have to select AI_THINKER mode to program and provide external 5V to make camera working. Current rating of FTDI is not sufficient to on the camera.

maheshvsnl123, 2020-01-29:
> this points to a problem with SIOD and SIOC lines. This is in reality Software I2C. Maybe add pull-ups to those lines?

**HAI this is problem with power supply ..use 5 volt opposite pin of 3.3 volt..you can use 3.3 for programming after removing all cable serial connector and connect 5 volt thats all.. get ip address then connect to google or application  **