espressif/arduino-esp32 #2465: Unwanted pulse with ledc

bdring, 2019-02-16:
When I use ledc in my program, I get an unwanted initial pulse on the pin. here is the simplest program I could write to cause the problem

```
#define PWM_CHANNEL 0

void setup() {
  ledcSetup(PWM_CHANNEL, 5000, 8); // setup the channel
  ledcAttachPin(GPIO_NUM_17, PWM_CHANNEL); // attach the PWM to the pin
  ledcWrite(PWM_CHANNEL, 0); // behaves the same if this is here or not
}

void loop() {
}
```

Here is the pulse on my logic analyzer.

[image]

What can be done to prevent this pulse? 

The PWM is used to control a laser in an engraver. This duration causes a big mark on the work piece.

lbernstone, 2019-02-16:
Does it happen if you write the channel duty to 0 before you attach the pin?

bdring, 2019-02-16:
There is no change in behavior if I move that.

stickbreaker, 2019-02-16:
Try pre configuring the pin

```c++
digitalWrite(pin,LOW);
pinMode(pin,output);
ledcAttachPin(...);
```

bdring, 2019-02-16:
No change with this code

```
#define PWM_CHANNEL 0
#define PWM_PIN GPIO_NUM_17

void setup() {
  digitalWrite(PWM_PIN,LOW);
  pinMode(PWM_PIN,OUTPUT);
  ledcSetup(PWM_CHANNEL, 5000, 8); // setup the channel
  ledcWrite(PWM_CHANNEL, 0); // behaves the same if this is here or not
  ledcAttachPin(PWM_PIN, PWM_CHANNEL); // attach the PWM to the pin  
}

void loop() {
}
```

[image]

bdring, 2019-02-16:
It looks like the **pinMode(...)** might be causing the problem. This program also causes the unwanted pulse.

```
#define PWM_PIN GPIO_NUM_17

void setup() {
  digitalWrite(PWM_PIN,LOW);
  pinMode(PWM_PIN,OUTPUT); 
}

void loop() {
}
```

Swapping the order of those two lines does not change anything.

bdring, 2019-02-16:
I think the problem is due to another part of my circuit. An output enable on a level shifter might not working right.

closed 2019-02-16 by bdring