espressif/arduino-esp32 #2466: Work WiFi.onEvent (WifiEvent) in loop function

DeltaVetal26, 2019-02-16:
Hi,

My code:
```cpp
bool wifi_connected; // Wifi Connect Flag

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_AP_START:
      //can set ap hostname here
      WiFi.softAP(AP_SSID, AP_PASS);
      //enable ap ipv6 here
      WiFi.softAPenableIpV6();
      break;
    case SYSTEM_EVENT_STA_START:
      //set sta hostname here
      WiFi.setHostname(AP_SSID);
      break;
    case SYSTEM_EVENT_STA_CONNECTED:
      //enable sta ipv6 here
      WiFi.enableIpV6();
      break;
    case SYSTEM_EVENT_AP_STA_GOT_IP6:
      //both interfaces get the same event
      break;
    case SYSTEM_EVENT_STA_GOT_IP:
      //wifiOnConnect();
      wifi_connected = true;
      break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
      wifi_connected = false;
      break;
    default:
      break;
  }
}

void setup() {
  WiFi.onEvent(WiFiEvent);
  WiFi.begin(wifiSSID, wifiPassword);
}

void loop() {
  if(wifi_connected) {
    // Ok. Wifi Enabled
  } else {
    // Wifi Connect Problem
  }
}
```

WiFi.onEvent () is in void setup (), and I have doubts, can I read Wi-Fi network conditions in a void () loop using the WiFiEvent function? It seems that the void setup() works in just one cycle. Or does it simply indicate the ESP to the desired function, which will be processed in a  void loop()?

lbernstone, 2019-02-16:
WiFi runs in its own task, so it continues after setup completes.  https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L99

pfeno, 2019-03-19:
@DeltaVetal26 @lbernstone 
This code is working?
When I copy/paste in arduino IDE, I have the following error 

```
error: variable or field 'WiFiEvent' declared void
 void WiFiEvent(WiFiEvent_t event)
                ^

error: 'WiFiEvent_t' was not declared in this scope
 error: variable or field 'WiFiEvent' declared void
 void WiFiEvent(WiFiEvent_t event)
                ^

error: 'WiFiEvent_t' was not declared in this scope
exit status 1
variable or field 'WiFiEvent' declared void
```

DeltaVetal26, 2019-03-19:
@pfeno 
Yes. Can you show the code completely?

DeltaVetal26, 2019-03-19:
@pfeno 
There is no 
```cpp
 #include <WiFi.h>
```
 library in this code.
Try to install it at the beginning of the code.

pfeno, 2019-03-19:
@DeltaVetal26 
On my business project, I need to add 

> #include <ETH.h>

to bypass 2 error:
- error: variable or field 'WiFiEvent' declared void
-  error: 'WiFiEvent_t' was not declared in this scope

lbernstone, 2019-03-19:
@DeltaVetal26 : Is your original issue answered?

DeltaVetal26, 2019-03-20:
@lbernstone , yes.

closed 2019-08-15 by stale[bot]