espressif/arduino-esp32 #2469: ESP32 OTA Update.h Not Writing To Correct Address

wezleysherman, 2019-02-17:
Hello, so I'm attempting to perform an OTA update via bluetooth and am running into some issues as far as writing to the correct partition.

Context:

I have my partition table setup as:
```
# Name,   Type, SubType, Offset,   Size, Flags
nvs,      data, nvs,     0x9000,   0x5000,
otadata,  data, ota,     0xe000,   0x2000,
app0, 	  app,	factory, 0x10000,  1150K,
firm,	  app,	ota_0, 	 ,		   2400K,
eeprom,   data, 0x99,    0x390000, 0x1000,
spiffs,   data, spiffs,  0x391000, 0x6F000,

```

app0 is where my updater code lies and I'm utilizing the Update.h lib to write chunks of data to memory.

I'm able to successfully perform `Update.write((uint8_t*)(&image), sizeof(image));` and verify there are no errors utilizing `Update.hasError()`. Upon completion I'm using `Update.end()`.

Everything on the update side seems to work, but when I go to set the boot partition I'm utilizing:
```
    PART = esp_ota_get_running_partition();
    PART = esp_ota_get_next_update_partition(PART);
    esp_ota_set_boot_partition(PART);
```

As soon as `esp_ota_set_boot_partition(PART);` is called I'm getting this error: `esp_image: image at 0x130000 has invalid magic byte`. I'm assuming this is because the partition I want to boot into is completely empty. Is there a way to explicitly state which partition Update.h writes to?

lbernstone, 2019-02-17:
Not sure why your OTA isn't working, as it seems like your partitioning should be valid.  Here is a snippet that should at least help you with your debugging:
```
#include <esp_partition.h>

void partloop(esp_partition_type_t part_type) {
  esp_partition_iterator_t iterator = NULL;
  const esp_partition_t *next_partition = NULL;
  iterator = esp_partition_find(part_type, ESP_PARTITION_SUBTYPE_ANY, NULL);
  while (iterator) {
     next_partition = esp_partition_get(iterator);
     if (next_partition != NULL) {
        Serial.printf("partiton addr: 0x%06x; size: 0x%06x; label: %s\n", next_partition->address, next_partition->size, next_partition->label);  
     iterator = esp_partition_next(iterator);
    }
  }
}

void setup() {
  Serial.begin(115200);
  partloop(ESP_PARTITION_TYPE_APP);
  partloop(ESP_PARTITION_TYPE_DATA);
}
void loop() {}
```

wezleysherman, 2019-02-17:
Thank you @lbernstone !

Okay so it is printing out the partition correctly.
`partiton addr: 0x130000; size: 0x25f800; label: firm`

I'm sending the raw hex from the compiled .bin and am writing it as a string casted to uint8_t.

I'm checking each chunk to make sure there are no errors.

Here is my code: `UPDATE_SIZE = 40000`
```

if (image.length() <= UPDATE_SIZE){
        image += input;
        return;
      }
      else {
        Serial.println(ESP.getFreeHeap());
        Update.write((uint8_t*)(&image), sizeof(image));
        image = ""; // dump string buf
       
        if (!Update.hasError()) {
          Serial.println("Wrote chunk!");
        }
        else {
          Update.printError(Serial);
          Serial.println("Corrupt chunk!");
        }
        return;
      }
```

This is within my bluetooth receive callback, which takes 256 characters at a time and saves them to the `image` string. Then once that string has 4000 characters it casts them to uint8 and writes them.

closed 2019-08-15 by stale[bot]