espressif/arduino-esp32 #2473: Protection against damage NVS storage?

DeltaVetal26, 2019-02-18:
Hi,

For storing Wi-Fi connection settings and some other data, I use NVS storage. Now I saw an [article](https://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/) stating that the NVS storage could be damaged, leading to a Wi-Fi crash.  My ESP module is in an inaccessible place, so I will not be able to get close to it if the NVS is damaged.

The question is, what can cause damage to the NVS and how to prevent it?

me-no-dev, 2019-02-18:
Nothing in general should cause such thing. If it happens then it's a bug :) NVS is used by the IDF subsystem. BTW that article is quite vague

DeltaVetal26, 2019-02-18:
Is it safe to store WIFI settings in NVS?
In the code, I only do this:

1) Write WIFI Settings
```cpp
 preferences.begin("wifi", false); // Note: Namespace name is limited to 15 chars
   preferences.putString("ssid", qsid); // Write SSID Network
   preferences.putString("password", qpass); // Write PASSWORD Network
   delay(300);
 preferences.end();
```

2) Read WIFI Settings
```cpp
  preferences.begin("wifi", false);
    wifiSSID = preferences.getString("ssid", "none"); // NVS SSID
    wifiPassword = preferences.getString("password", "none"); // NVS Password
  preferences.end();
```

me-no-dev, 2019-02-18:
wifi settings are automatically stored in NVS when you WiFi.begin(ssid, pass); The above is not necessary :)

closed 2019-08-15 by stale[bot]